motioneye-project/motioneyeos #2806: Light Switch Detection completely useless

feraldruid22, 2021-09-11:
Now I understand why is this feature disabled by default. By not changing any other settings if I turn this on and can put it to any percentage 10 20 30 ... 100 then the camera will not detect motion anymore.

I'm not even sure if this feature was intended to do what I need which is:

Like 5-10 times a day when clouds come and go the overall brightness in the apartment changes which turns the camera on like somebody would be there.

Currently I using frame change treshold 13%.

The camera is indoor 360 degree fisheye and does not look to any windows directly.

Anyone got tips how to adjust this?

I have also tried this Despekle filter, same thing if I turn it on motioneye does not record anything anymore.

starbasessd, 2021-09-11:
Despeckle basically removes noise in the video (on some of my cameras most noticeable at dawn and dusk) that causes false positives for motion detection.
The settings I use are 
Frame Change Threshold 0.1%
Auto Threshold Tuning Off
Auto Noise Detection On
Despeckle Filter On

I have 2 Foscam cameras that I use Maximum Frame Change instead of Light Switch for 'sudden' light changes. 
Even then, cloud coverage isn't all that fast and still gives 'false' positives.
Another interaction that may be affecting detection is Motion Gap and Minimum Motion Frames. I have mine set to 2 each. THAT cuts a lot of false positives, along with the Frame Change Threshold 0.1% above.

A lot of the settings interact with each other. I find it helpful to turn on Show Frame Changes while setting up the cameras, then turn it off until/unless diagnosing false positive or negative triggers.

starbasessd, 2021-09-11:
Also, in the future, please answer the questions when opening an issue. It is very helpful for diagnostics and recreating the issues, as some combinations of boards, version of software, and cameras have known issues.

closed 2021-09-11 by starbasessd

feraldruid22, 2021-09-13:
This was actually a bad advice, I have tried the settings you have mentioned and the camera turned on like 120 times for everything.

starbasessd, 2021-09-13:
As stated, the settings are interactive. Start with the defaults, and adjust to fit your situation and hardware. It may take a while. The light switch detection does work in mine. Since you still refuse to answer the original questions: Version of motionEyeOS, make and model of board, make and model of camera, how connected to network, any logs, etc, I locked this issue. Please open a new issue, with those at a minimum, and your motion detection settings.