motioneye-project/motioneyeos #2807: Can't see live preview feed, even though it was working at one point

ayrfield2, 2021-09-12:
I had trouble with my camera a while ago, so i did a fresh install today.
Initially, the camera was working - i could see the live feed. I changed a few minor settings and now i am just getting the icon of the camera with the line through it.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: ZeroCam Fisheye NightVision - for PiZero & Raspberry Pi 3

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

n/a

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7150351/motioneye.log)

starbasessd, 2021-09-12:
What specifically: " I changed a few minor settings "
Please include the other 4 logs. (Under Settings, Expert Settings. motion.log, messages.log, dmesg.log, and boot.log)
If you can console in, or SSH in, what is the result for:
vcgencmd get_camera
If you would like to reset things, you can delete the /data/etc/camera-1.conf and reboot

ayrfield2, 2021-09-14:
Minor settings were were setting up the wifi and trying out resolutions and
frame rates to see what looked good.
I will try that command as soon as I get a chance.

---- booting motionEyeOS 20200606 ----
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.83-v7 ***@***.***) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018.11 (arm-rel-8.26))) #1 SMP Sat Jun 6 14:14:33 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] On node 0 totalpages: 196608
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xb0/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36684 r8192 d20660 u65536
[    0.000000] pcpu-alloc: s36684 r8192 d20660 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 756380K/786432K available (8192K kernel code, 655K rwdata, 2228K rodata, 1024K init, 820K bss, 21860K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
                   lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 656 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 821 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26874 entries in 79 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000281] Console: colour dummy device 80x30
[    0.000300] console [tty1] enabled
[    0.000341] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000359] pid_max: default: 32768 minimum: 301
[    0.000678] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000693] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001623] CPU: Testing write buffer coherency: ok
[    0.002084] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002735] Setting up static identity map for 0x100000 - 0x10003c
[    0.002885] rcu: Hierarchical SRCU implementation.
[    0.003679] smp: Bringing up secondary CPUs ...
[    0.004524] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005463] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006326] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006449] smp: Brought up 1 node, 4 CPUs
[    0.006461] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006467] CPU: All CPU(s) started in HYP mode.
[    0.006472] CPU: Virtualization extensions available.
[    0.007410] devtmpfs: initialized
[    0.020333] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.020572] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020592] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021191] pinctrl core: initialized pinctrl subsystem
[    0.022043] NET: Registered protocol family 16
[    0.025058] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.030708] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 885 lines cut ...]
[    0.713127] [vc_sm_connected_init]: start
[    0.721660] [vc_sm_connected_init]: end - returning 0
[    0.723070] Initializing XFRM netlink socket
[    0.723095] NET: Registered protocol family 17
[    0.723210] Key type dns_resolver registered
[    0.723717] Registering SWP/SWPB emulation handler
[    0.724358] registered taskstats version 1
[    0.724374] Loading compiled-in X.509 certificates
[    0.731575] uart-pl011 3f201000.serial: cts_event_workaround enabled
[... cut ...]
[    0.799504] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    0.802346] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    0.807715] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    0.809021] of_cfs_init
[    0.809175] of_cfs_init: OK
[    0.809849] Waiting for root device /dev/mmcblk0p2...
[    0.873905] random: fast init done
[    0.895275] mmc0: host does not support reading read-only switch, assuming write-enable
[    0.899434] mmc0: new high speed SDXC card at address aaaa
[    0.900631] mmcblk0: mmc0:aa

starbasessd, 2021-09-14:
Adjusting the resolution and frame rate to something that motionEye/motion/ffmpeg and your camera can't agree on as well as the frame rate can cause the gray screen. In that case, try setting the resolution to 640x480 and frames to 5fps and then slowly move it back up from there.

ayrfield2, 2021-09-15:
I tried resetting the resolution and no improvement.
I also tried SSH-ing in and running the command but there was no response.
Maybe I'm just not familiar with the protocol but it looked like there was
no response to any commands. I couldn't even navigate directories.

starbasessd, 2021-09-15:
Sounds like a candidate for a re-install from scratch.

ayrfield2, 2021-09-15:
This was just freshly installed over the weekend. I'll try spin it up again
but I would be concerned that the same problem could crop up if I don't
know what caused it in the first place.

starbasessd, 2021-09-15:
Understood, however, making random changes that you can't document makes it hard to figure out what 'broke' in your setup. I try to reproduce other's issues here in my environment. We just discovered an obscure bug after the user documented specifically what he was naming his cameras. I was then able to reproduce the issue. 
When you re-image, please make one change at a time, so we can see where it breaks. It may be helpful to do a back up at each stage/step so you don't have to rebuild from scratch every time.
At the point of breakage, if you can save the 5 log files (boot.log, motion.log, motioneye.log, messages and dmesg, then zip and attach them to your report, it would be tremendously helpful, too. (They are found under Settings, Expert Settings)

realdeadbeef, 2021-09-18:
I've just installed MotionEyeOS and as soon as I load the web UI it has the picture of a camera with a line through it, I have a NoIR cam connected and when I remove to camera it just breaks and says that I need to refresh. Then the UI goes down and i can't access the interface

starbasessd, 2021-09-18:
@realdeadbeef Please open a new issue with your specifics, board, OS version, camera, etc. so we can help you.