motioneye-project/motioneyeos #2808: no more as 4 cams?

voicenoise, 2021-09-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20200606).

### Board Model

I am using the following board/model: (Raspberry PI 4 -8GB).

### Camera

I am using the following type of camera: (Network Camera and Simple MJPEG Camera).
My camera model is: (not necessary).

### Network Connection
My motionEyeOS unit is connected to the network via: (Ethernet).

I like this project and hope still to come few updates like booting from raspi 4 8GB
but this is not my issue here...

i try 3 times and every time are the same, after adding a fifth network cam the system hangs and cannot reboot.
failed at boot by Starting motioneye.
no matter what a cam rtsp or mjpeg

starbasessd, 2021-09-13:
For the Pi4-8GB use dev20201026.
Once you are confident it's working for you, you can do a backup, then install and boot to USB
Change cmdline.txt from root=/dev/mmcblk0p2 to root=/dev/sda2
When booted up, you should be able to do the restore of your camera configs.

voicenoise, 2021-09-13:
i try dev20201026 on USB and restore from a backup with a 4 cam configuration.
Here are the same he boots up to
* Starting motioneye:
then hangs a few minutes then comes "failed" and the device rebooting.

i try now a fresh installation with dev20201026 o USB and look what happening here after adding the fifth cam...

starbasessd, 2021-09-13:
Here I am with 6 cameras, 3 d-Link Network, 2 Foscam RTSP, and a generic Tiga USB Webcam...
on a Pi4-8GB on a USB boot (128GB) with dev20201026, both ethernet & wifi connections.
[image]

starbasessd, 2021-09-13:
What becomes 'failed' during boot? Post all 5 logs, especially boot.log and motioneye log.

voicenoise, 2021-09-13:
nice, mine wont work

also here with dev20201026 on USB

after click ok to add the 5 cam in this case a mjpeg cam 
(but it also happens with a rtsp connection)
come a short warning in the web interface 
"An error occurred. Refreshing is recommended."
and all streams from the other cams are down

after refresh and new login:
"You have not configured any camera yet. Click here to add one... copyright © Calin Crisan An error occurred. Refreshing is recommended."

Then i try directly the new backup that i have done before adding the fifth cam.
* Starting motioneye hanging then failed and the bootlop begin...

How do I get the logs files if I can no longer boot properly?

starbasessd, 2021-09-13:
Can you SSH in, or use a keyboard & monitor?
If not then:
Something got seriously corrupted.
Start with a fresh image, either on USB with the change above to cmdline.txt or on an SDCard.
Go to the WebGUI
Pick one of your cameras
Add it 5 times. Yes, the same camera info for all 5 add cameras.
Does it still crash?

voicenoise, 2021-09-13:
[log.zip](https://github.com/ccrisan/motioneyeos/files/7157498/log.zip)
i found a way to get the log files
i try with 5x same cam....

voicenoise, 2021-09-13:
it behaves slightly differently and does not crash. but one now event to streams are grey.
[image]
[image]
[image]

Reboot working too... so perhaps it is that mjpeg cam

starbasessd, 2021-09-13:
So delete all 5 cams, and pick another camera to do it again. Maybe the one you were adding last before.

voicenoise, 2021-09-13:
i can't add the mjpeg cam 5 times because no more streams on this cam, it is a older one.
i try to add this as first then the others 4 are the same model trendnet tvip 310 and another is a dlink dcs522l but this one i not added yet.

starbasessd, 2021-09-13:
You use the same entries 5 times same URL for all 5 cameras. For example my d-Link DCS5020L is 
http://IP_Address:80/mjpeg.cgi so I enter that for all 5 new cameras (plus user & password)

Once you have that set up, delete the cameras one at a time, and add another camera. Once it's all set up (still haven't changed ANYTHING ELSE) then do a backup. THEN start making other changes.

voicenoise, 2021-09-13:
all 6 cam was running now then i deactivate ftp service > reboot
the system hanging on 
*Starting motioneye: failed 
and the bootlop begins again 🙈

starbasessd, 2021-09-13:
So it has to do with the FTP service... For now, don't disable it, I'll look at it later.

voicenoise, 2021-09-13:
ok, i investivate now in folder /etc  
and recongnize that the files proftpd.conf and no_S61proftpd ha zero bytes

starbasessd, 2021-09-13:
Look in /data/etc, and /etc is read-only, changes are in /data/etc

voicenoise, 2021-09-13:
there is only one data folder in this 3 partitions and this one is completly empty

starbasessd, 2021-09-13:
try commands:
df -h
df -T
motionEyeOS uses 3 partitions, overlay file systems and read-only file systems
If /data is empty, the file systems are broken and you need to re-image, apply the backup, and do not change the FTP server settings.

voicenoise, 2021-09-13:
I'm already at it. from scratch again 🙈

voicenoise, 2021-09-13:
Backup is not working although i create the backup before deactivate the ftp service... 
*Starting motioneye: failed

very strange and i give up for today

voicenoise, 2021-09-14:
So when I change the cam name and click reboot the error comes back.
i have not strange names: CAM1 , Frontview, Outdoor, Room etc.

starbasessd, 2021-09-14:
Do you have any non-alphabetical/numerical characters in the names anywhere? Punctuation, quotes, single quotes, mathmatical, etc?

voicenoise, 2021-09-14:
no as i said only like this one
> i have not strange names: CAM1 , Frontview, Outdoor, Room etc.

voicenoise, 2021-09-15:
i can try with the raspi soon, because every time reinstall needs a lot time.

but i have testing with a ubuntu 20.04.3 Machine
and running into a similar thing.
after try to add the DLink DCS-522L i got:

"An error occurred. Refreshing is recommended."
and all streams from the other cams are down

after refresh and new login:
"You have not configured any camera yet. Click here to add one... copyright © Calin Crisan An error occurred. Refreshing is recommended."

reboot the machine = service status failure:

motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-09-15 02:09:20 CEST; 1s ago
    Process: 2168 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 2168 (code=exited, status=1/FAILURE)

Sep 15 02:09:19 motion-eye meyectl[2168]:     _get_additional_config(main_config)
Sep 15 02:09:19 motion-eye meyectl[2168]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2020, in _get_additional_config
Sep 15 02:09:20 motion-eye meyectl[2168]:     (sections, configs) = get_additional_structure(camera=bool(camera_id))
Sep 15 02:09:20 motion-eye meyectl[2168]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1991, in get_additional_structure
Sep 15 02:09:20 motion-eye meyectl[2168]:     result = func()
Sep 15 02:09:20 motion-eye meyectl[2168]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
Sep 15 02:09:20 motion-eye meyectl[2168]:     import pytz
Sep 15 02:09:20 motion-eye meyectl[2168]: ImportError: No module named pytz
Sep 15 02:09:20 motion-eye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Sep 15 02:09:20 motion-eye systemd[1]: motioneye.service: Failed with result 'exit-code'.

try delete the new last cam settings
sudo rm /etc/motioneye/camera-7.conf

but not success

so next i do:

sudo systemctl stop motioneye

delete directories:
sudo rm -rf /etc/motioneye && sudo rm -rf /var/lib/motioneye

recreate directories:
sudo mkdir -p /etc/motioneye && sudo mkdir -p /var/lib/motioneye

copy configs:
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

start motioneye:
sudo systemctl restart motioneye

interface working now again

but import the last backup fails again (cams renamed)
"You have not configured any camera yet. Click here to add one... copyright © Calin Crisan An error occurred. Refreshing is recommended."

So i do it the above again and take the first backup with cam not renamed

> working again

very very strange

Cam names are: CAM1 ; Outdoor ; Frontwiev ; Room ; Eingang ; 95

voicenoise, 2021-09-15:
last night i used the backup then it worked. This morning it hanging and after do a reboot the service is gone again:

● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-09-15 11:58:01 CEST; 11min ago
    Process: 795 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 795 (code=exited, status=1/FAILURE)

Sep 15 11:58:00 motion-eye meyectl[795]:     _get_additional_config(main_config)
Sep 15 11:58:00 motion-eye meyectl[795]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2020, in _get_additional_config
Sep 15 11:58:00 motion-eye meyectl[795]:     (sections, configs) = get_additional_structure(camera=bool(camera_id))
Sep 15 11:58:00 motion-eye meyectl[795]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1991, in get_additional_structure
Sep 15 11:58:00 motion-eye meyectl[795]:     result = func()
Sep 15 11:58:00 motion-eye meyectl[795]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
Sep 15 11:58:00 motion-eye meyectl[795]:     import pytz
Sep 15 11:58:00 motion-eye meyectl[795]: ImportError: No module named pytz
Sep 15 11:58:01 motion-eye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Sep 15 11:58:01 motion-eye systemd[1]: motioneye.service: Failed with result 'exit-code'.

i come from zoneminder and i really like this system also the resources and power consumption are much better than zm. 
But is not stable enough for me at the moment :-(

ozett, 2022-06-16:
give this a try: https://frigate.video/

have it running on an old intel i5 with 12 cams.
but you need an AI accellerator for 40$ (hard to buy at the moment)