motioneye-project/motioneyeos #2810: Punk kids on my lawn

Ventnts, 2021-09-21:
Hi,

I have a situation where there some low life children throwing rocks 150 feet down to my kids and family.  They have not come back in a long time, but I want to use Cris's SW to be able to detect someone doing something, like throwing rocks!

I know he can detect frame changes, but I feel like we can add some open source AI smarts to detect certain movements.

Is anyone game for this?  I would like to help figure it out, or just do it.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

skycarl, 2021-10-02:
Would something like Frigate serve your use case? https://github.com/blakeblackshear/frigate

starbasessd, 2021-10-17:
@sirjeannot What is crashing? 
Anything in logs? 
Have you checked temperatures and voltage. 
Are you running as Fast Network Camera ? \( Assuming yes, since you mention streaming to Frigate using rtsp \) 
Which version of motionEyeOS?

sirjeannot, 2021-11-21:
oh no, overclocking is out of question. it's all set to default voltage (no parameter in config.txt), and even went downclocking the cpu. btw, the very same raspi0 not connected to the csi camera doesn't crash after hours of cpu load, with the same config.txt.
so as far as I'm concerned, I'm ruling out this type of hardware.

starbasessd, 2021-11-21:
When you are up, ssh in and report results for
vcgencmd measure_temp
vcgencmd get_throttled