motioneye-project/motioneyeos #2812: Can’t store files from different cameras in different folders

arielylim, 2021-09-22:
I'm not sure if I am the problem here but I can't do a network share for a different folder for my different cameras. 

I'm running motioneyeos on a pi 4B via docker. 

I have a pi0 (let's call this cam1) running motioneyeos that stores via samba to my 4B. All working fine. 

I just added a new pi0 (cam2), same settings. It works with a caveat: i can only store the recordings together with the other pi0. 

Meaning, when i setup the storage path, I can only use the same file path [samba-share]/motioneye/Camera1. 

I want to add it to [samba-share]/motioneye/Camera2 but im getting an error "accessing network share failed: cannot mount network share"

In my samba config, the entire folder of /home/pi/samba-share is what i used. So i dont see any reason why it cannot access the network share. Is there a special config i need to do in motioneye?

Again, if i chose the same network path as my camera1, it works. I can see the recordings of both cam1 and cam2 but it's mixed together. That's why i want to split it so it's easier to manage and look at later on. 

TIA

arielylim, 2021-09-22:
Sorry, i just had a breakthrough. 

Apparently i had a spelling error in my camera2 folder. Fixed it,tried the test share, and it's working now. 

✌️

starbasessd, 2021-09-22:
Issue #1, the docker is motioneye, not motioneyeOS, you'd be better looking at 
https://github.com/ccrisan/motioneye
That said, how are you running the docker container? docker run command from:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker?
Many more questions, but let's get this one first.

closed 2021-09-22 by starbasessd