motioneye-project/motioneyeos #2813: Invalid credentials

Learningasig0, 2021-09-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

motioneyeos-raspberrypi3-20200606

Hi,
 I seem to be having a issue recently with my Authentication originally when i set everything up earlier this month i had no issues but the past couple days i have made no modifications yet my password has become incorrect, so i decided to do a fresh image as i dont own a linux based computer to check logs (windows 10) after doing a fresh reset with no cameras attached i connected it to my lan and powered it up to modify my wifi network. After applying the settings to my wifi and adding password and clicking apply my creditials have became invalid,
 
 I have read previous threads involving this issue and tried the following steps
 5 ***** 
 random generated passwords in other threads 
 admin, Blank 
 
 so far i have had no success if anyone has any ideas besides doing another fresh install please let me know

starbasessd, 2021-09-25:
Unfortunately, if the password isn't working, you can't get to the files you'd need to edit. 
(/data/etc/motioneye.conf)

Please try dev20201026.

Learningasig0, 2021-09-25:
> Please try dev20201026.

Whats is this?

starbasessd, 2021-09-25:
The latest version. For the Pi3:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz
Over the past 11 months it has been proven to be a bit better than 20200606, especially on the Pi4s, but there are known issues that it resolves, too, like continuous recording. You could also down grade to 20200203, but there are a number of issues with it resolved in 20200606 and dev20201026.