motioneye-project/motioneyeos #2814: Is there a features roadmap?

xavi-developer, 2021-10-03:
Hello,

first of all, congratulations for this great project. I was going to start something similar but this project is awesome.

Apart from what the project currently offers, I was wondering about some features such:
* More type of sensors (door / window opening, volumetric, presence, light...)
* Receive a notification in a telegram group when some sensor is triggered or be able to notify the group through the email 

I don't know if there are already planned features like those. I am a 20-year-experienced developer and I would be willing to contribute with the development of it.

Thank you!
Xavi

Ventnts, 2021-10-03:
I hear you man. He put in hooks that allow us to do anything when it's triggered.  This would include the GPIO.  Do you know where they are?

I tried having it connect to Google s SMTP server and used my TMobile text email. It was about a 5-10 second delay but it was cool.  

I got a GPIO kit to play with loud and quiet buzzers and other tthings.

Lastly I love his work too.   I would like to be able to make a mini AI  worker to identify movements but I just want to know if someone is throwing something.  @starbasessd helped me find a project that's doing that.

starbasessd, 2021-10-03:
Python & Python GPIO are there.
https://github.com/ccrisan/motioneyeos/wiki/Tweaks
You can trigger things (in or out) using Settings, Motion Notifications or
Settings, File Storage which gives you the equivalent of a 'plug in' functionality,
or use Action Buttons for manual triggers
https://github.com/ccrisan/motioneye/wiki/Action-Buttons
and you can put any script or executable in place with those names and they are per-camera.
Several people have posted triggers for Telegram. 
As to sensors, due to the literal bazzillions of them out there with i2c, spi, gpio and other ways of communicating, other than putting in the python capability to talk to any of them, what more would you want?

starbasessd, 2021-10-03:
BTW, I wrote a 'quick and dirty' script a long time ago and posted in the motioneye github issues, to trigger playing an MP3 file as a doorbell. Turned any camera used into a NEST or RING presence notifier. Not a big deal. I am not a programmer/developer by any stretch of the imagination (I do Help Desk Support at the Enterprise Level)  but it's not hard to write such a script in python, or even bash. (Harder in bash in motionEyeOS because it uses busybox for almost everything).

Ventnts, 2021-10-03:
@starbasessd thank you for the tips and I will read more, it sure saved me time.

xavi-developer, 2021-10-04:
wow, that's a lot of possibilities, I'll take a further look at them and check what can be done

Thank you very much

closed 2021-10-10 by xavi-developer