motioneye-project/motioneyeos #2815: Raspberry Pi Upgrade Question

Dewey3, 2021-10-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:  20190911

### Board Model

I am using the following board/model:  Raspberry Pi 3B & Raspberry Pi 3B+

### Camera

I am using the following type of camera:  V4L2

My camera model is:  USB 1MP IR Dome Cameras

### Network Connection

My motionEyeOS unit is connected to the network via:  WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:  N/A

### Log Files

I consider the following log files relevant to this issue:  N/A
 

Greetings all!  While this is not an issue, it is a question completely and directly related to MotionEyeOS.  I am running 15 individual Raspberry Pis, each connected to its own USB 1MP IR camera.  While each Pi/camera is totally self dependent, all of the Pis/cameras are linked to a "central" headless Raspberry Pi that is used for accessing the cameras.  Some of the first Pis I installed are now crossing the 5 year old mark, so I am considering upgrading them from Pi 3Bs (before the Plus model) to 4s.  From everything that I have seen the Raspberry Pi 4's processor is only .3 MHz faster than the 3B (and only .1 faster than the 3B Plus), but the GPU appears to be twice as fast, which leaves me to wonder how much of an "upgrade" would I really be getting.  Does anyone have the knowledge on the internal workings of the Raspberry Pis & MotionEyeOS to say if there would be a significant end-user gain in upgrading from the 3B (and later 3B Plus) to Raspberry Pi 4s?  Also, can MotionEyeOS make better use of the larger memory of the 4s since they come in 2GB, 4GB and 8GB?  I know I will have to upgrade from the 20190911 MotionEyeOS version that I'm using to dev20201026 which doesn't bother me, I just want to make sure that upgrading perfectly functioning Raspberry Pi 3s to Raspberry Pi 4s would be a sound decision.  Thanks all.

starbasessd, 2021-10-04:
Interesting. I safely overclock my Pi4b-4GB and 8GB to 2.0GHz on a regular basis. Overall, though, performance is still only a bit better. There are a lot of tweaks you can do to improve processing, but motion (the program doing most of the heavy lifting) hasn't been updated to take advantage of the greatly extended RAM of the 4Bs. I wish they could get the best optimization from the GPUs too, for encoding.
But if I were doing it, I'd go for best bang for the buck. If I could find Pi3B+ or even Pi3Bs at $20 each, I'd go that way. I'm sure that it would be handy to have a few spares around, if they've reaching the 5 year old mark. Investing in a few new power supplies, maybe some better cameras... The Official PiCam HQ is nice, and with that and motionEyeOS, can be used in Fast Network Camera mode, giving you a much better performance improvement.

Dewey3, 2021-10-04:
Thank you Kevin, and thank you for all of the support that you provide here for us!  My first thought was to upgrade to 3B+ models since I would at least be getting 5G wifi over the pre-5G B models, which is just over half of my stock.  However, not including any sales or specials, the 3B+ and 4 - 2GB are running at the same starting price, which is why I'm having a hard time accepting upgrading to 3B+.  I have been keeping 1 & 2 spares on hand for malfunctions, but I always seem to find somewhere to use them, which includes the 14th camera which just sits in a window sill and gives a live view going down our street.

Also, thank you for the upgraded camera tip, but for now, the dome cameras that I have been using have really proven their worth in lasting those years on outside walls through our summers and winters.  I will be upgrading some of the Pis ... eventually ... I just have to finish the fight in my head over upgrading to 3B+ for the immediate 5G improvement versus upgrading to the 4, which will require new cases (a little more cost to upgrading) and hoping that a future release will harness the power of the faster GPU.

Dewey3, 2021-10-04:
Thanks again!

closed 2021-10-04 by Dewey3