motioneye-project/motioneyeos #2822: Cannot connect via RTSP

mlittle408, 2021-10-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W.

### Camera

I am using the following type of camera: PI Camera v2.

My camera model is: V2

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

### Peripherals

None

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

None

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I am running MotionEyeOS in Fast Network Camera mode. I am not able to connect via rtsp://192.168.1.154:554/ or rtsp://192.168.1.154:554/1 or rtsp://192.168.1.154:554/Camera1. I have tried with VLC running on Windows 10

As a point of reference, I am able to connect via HTTP when not running Fast Network Camera so I know the camera is working. 

Did I miss a step somewhere?

[messages.log](https://github.com/ccrisan/motioneyeos/files/7365889/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/7365890/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/7365891/dmesg.log)
[streameye.log](https://github.com/ccrisan/motioneyeos/files/7365892/streameye.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7365893/motioneye.log)

starbasessd, 2021-10-18:
There are a number of issues with motionEyeOS 20200606. Can you please try dev20201026?

mlittle408, 2021-10-18:
I did the update, rebooted and now the left admin panel is all greyed out. 

Also, this is the URL for RTPS? http://ip_address:8081/h264

starbasessd, 2021-10-18:
At the top left, click on the head bust icon, and log in as admin.

Yes, the RTSP URL is http://ip_address:8081/h264