motioneye-project/motioneyeos #2826: Camera grayed out

peterd550, 2021-10-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Yes I have
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes I have
### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
0.42.1
### Board Model
Raspberry PI 4 - 4GB

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera
Network Camera
D-Link dcs-5020L - but with a lag
Tapo camera c100

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
Network Camera

My camera model is: (insert your camera model, e.g. Logitech C290).
D-Link dcs-5020L - but with a lag
Tapo camera c100
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
WIFI
### Peripherals

I am using the following peripherals that I consider relevant to this issue:
Nope
 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/7406493/boot.log)
[motioneye.log](https://github.com/ccrisan/mo
[motion.log](https://github.com/ccrisan/motioneyeos/files/7406496/motion.log)
tioneyeos/files/7406495/motioneye.log)

peterd550, 2021-11-08:
Resolved issue. Don't know how just start working
I now have 2 more all working as expected

closed 2021-11-08 by peterd550

starbasessd, 2021-11-08:
It happens. Sometimes it's the phase of the moon, or whether someone is holding their tongue 'just right'...

peterd550, 2021-11-08:
Hahaha yep I know that feeling.
Thanks for all the support and knowledge