motioneye-project/motioneyeos #2830: No image/movie saving

wahur-nan, 2021-10-31:
Hi there,

I am running motionEyeOS version: motionEyeOS 20200606
Board Model: Raspberry PI 3B+
Camera: MMAL
My camera model is: Raspberry Pi NoIR Camera Board v2 - 8MP
32 GB SD card storage with about 29 GB free space

With this config everything is working fine: we can see the pictures of the live stream on its website. 
The only problem is the saving of the pics and videos.
I tried to change the location to /tmp, but did not save anything to the card. I tried to change to /data/output but did not saved...
I also tried to check the relevant log files but I was unable to see any useful data.

What can be wrong?
Can somebody help to solve my issue?
Thank you,
Wahur

starbasessd, 2021-10-31:
The default path for saving locally for motionEyeOS is:
/data/output/Camera1
for the first camera. 
The system uses Read Only for most of the file system (/boot, /) to keep the system safe,
and Overlays (which point into the /data partition file tree)  in the paths, which is why 
/tmp doesn't work. Changes to /etc config files are made in /data/etc which over-ride
(overlay) the default /etc files, and so forth.
There are ways to change the default storage (if you have a USB stick inserted for external storage, for example) but the path must be real. You can also use a network path if to a SMB / CIFS share there, too, or use the Settings, File Storage, and turn on 'Upload Media Files', which gives you the ability to use FTP, SFTP, Google & Dropbox.
[image]

wahur-nan, 2021-11-02:
Thank you for your answer. It seems I messed up something and will start it over again. :)

closed 2021-11-02 by wahur-nan