motioneye-project/motioneyeos #2831: Can't connect to Pi Zero W when connected on wifi, works wired

Jayden933, 2021-11-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Raspberry Pi Zero W v1.1.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Arducam OV5647 Camera Module V1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

Unable to get any

I'm having the same issue as described in these two posts, but the solutions for them were unsuccessful for me.
https://github.com/ccrisan/motioneyeos/issues/2558
https://github.com/ccrisan/motioneyeos/issues/2764

I provided a wpa_supplicant.conf file and the Pi Zero successfully connects via wifi. I have it connected to a display and can see its output and everything works properly. But I can't reach the web UI for motioneyeos on the IP that the Pi Zero takes. On its first bootup, it was assigned 192.168.1.130. I get no response from that IP when I put it in my web browser, and when I try to ping it from my Windows 10 PC, I get the same "Reply from 192.168.1.65: Destination host unreachable." message that the users in those other two posts had. My Windows 10 PC's IP is the 192.168.1.65. From my router, I can see the Pi Zero connected at the correct IP address, 192.168.1.130 and it shows that it's connected and communicating with it. I don't think I need to give it a static IP, but I tried it as described in the other posts just to be sure. I first reimaged the Pi with a static_ip.conf with the contents:
```
STATIC_IP="192.168.1.130/24"
STATIC_GW="192.168.1.254"
STATIC_DNS="192.168.1.254"
```
 
Note, these values are the defaults that were assigned when it first booted, which should be correct. When I do `ipconfig` locally on my PC, it shows the default gateway as 192.168.1.254 (which is also the address where I can view/adjust router and network settings). Perhaps unsurprisingly, that didn't change anything, I still couldn't connect to the web UI from a browser and couldn't ping it. Though on bootup, the output did display that those values were static. I went into my router settings (which still saw the Pi Zero just fine at 192.168.1.130) and changed the Pi Zero's IP allocation to static at that address. Rebooted, and still can't connect.
I then tried reimaging with a static_ip.conf file with the contents:
```
STATIC_IP="192.168.1.130/24"
STATIC_GW="192.168.1.1"
STATIC_DNS="192.168.1.1"
```
just to see if it worked. When it booted this time, it accepted those values, but as a result could not connect to the internet (it wasn't able to get the date/time which it had done instantly before) and I still couldn't communicate with it.

I tried plugging a USB to ethernet adapter into the Pi Zero, hardwired it to the router, and rebooted. This time, for the wired connection it was assigned 192.168.1.73. Going to that address in the browser opened it immediately and works as expected. I just want to figure out how to connect to it on wifi. Any help would be greatly appreciated!

starbasessd, 2021-11-03:
What kind of WiFi are you using? I have heard of this issue with Mesh type Access Points. 
The fact that you are able to connect no issues with the USB Ethernet adapter indicates the issue is with your WiFi network.
I would also consider trying the RPiZero dev20201026 image:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz
Another thing to look at would be:
STATIC_IP="192.168.1.130/24"
STATIC_GW="192.168.1.1"
STATIC_DNS="192.168.1.1"
Shouldn't the GW and DNS be pointing to 192.168.1.254?

Jayden933, 2021-11-03:
Yes, it should be 192.168.1.254, but just thought I would try 1.1 since that was suggested in another thread (and I'm not super knowledgeable with networking stuff, unfortunately). 

It potentially could be my access point, because it is an ARRIS BGW-320 which is AT&T's new fiber ONT/router combo. But I don't think it's set up as a mesh wifi network, it's the only wireless access point I've got. But I'm not 100% sure how to check that either. I do have the 2.4 GHz and 5GHz bands separated out as individual wifi networks if that's what you mean, and I'm having the Pi connect to the 2.4 network.

I will try the newer dev build out and see if I can look at some router settings and report back later today!

Jayden933, 2021-11-03:
The dev20201026 build worked! Sort of... It worked in that I gave it the same wpa_supplicant.conf and the ip_config.conf with
```
STATIC_IP="192.168.1.130/24"
STATIC_GW="192.168.1.254"
STATIC_DNS="192.168.1.254"
```
and it connected to the network normally. And now I can open the web UI at 192.168.1.130! Which is great, except my intention is to add this as a secondary camera with my Raspberry Pi 3 (with its own camera) acting as a base station at 192.168.1.129. The Pi 3 is hardwired next to the router so I never had any issues with it since I was always going ethernet with it. But when I try to add a camera in the web UI for the Pi 3, I select Remote motionEye Camera and give it http://192.168.1.130/, it says "Failed to connect to 192.168.1.130 port 80: No route to host". So it seems now I'm at least able to connect to both Pis at .129 and .130, but it seems like they can't communicate with each other :(

starbasessd, 2021-11-03:
Try adding a Network Camera on the Pi3, using the URL of http://192.168.1.130:8081.

Jayden933, 2021-11-03:
same issue unfortunately
`Failed to connect to 192.168.1.130 port 8081: No route to host`

starbasessd, 2021-11-03:
Interesting. Try a different port on the router for the Pi3?
What do you get if you run the command 
traceroute 192.168.1.130
on the Pi3?
Can you ping the PiZero from the Pi3?
Can you ping the Pi3 from the PiZero?

Jayden933, 2021-11-03:
I used putty and ssh'd into each of them and tried to ping the other. Both times, it would say
`PING 192.168.1.130 (192.168.1.130): 56 data bytes`
and then just hang forever until I eventually hit ctrl+c and it said
```
--- 192.168.1.130 ping statistics ---
62 packets transmitted, 0 packets received, 100% packet loss
```
and vice versa from 130 to 129.

I tried traceroute 192.168.1.130 (PiZero) when ssh'd into 129 (Pi3), and got
```
traceroute to 192.168.1.130 (192.168.1.130), 30 hops max, 38 byte packets
 1  unknownXXXXXXXXXXXX.attlocal.net (192.168.1.129)  3116.253 ms !H  3111.597 ms !H  3119.917 ms !H
```
(I didn't know if the hex values were some kind of identifying information, that's why it's X'd out)

starbasessd, 2021-11-03:
Check your settings on the Pi3. 
ip a
and post what it has for eth0. specifically the inet and brd...
then run
route 
and post what it says for eth0 
for example:
pi@raspi1:~ $ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         router2         0.0.0.0         UG    202    0        0 eth0
I only need the line that starts default and ends eth0

Jayden933, 2021-11-03:
The eth0 section contains
```
link/ether b8:27:eb:d1:88:ac brd ff:ff:ff:ff:ff:ff
inet 192.168.1.128/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
```
and a couple other lines.

The line in question for route is
`default         dsldevice.attlo 0.0.0.0         UG    0      0        0 eth0`

starbasessd, 2021-11-03:
Your Pi3 is on 192.168.1.128, not 129...

starbasessd, 2021-11-03:
So, if Pi3 is on 128, and PiZero is on 130, then on the PiZero 
ping 192.168.1.128
traceroute 192.168.1.128
If they are no longer on static addresses then we have a whole 'nuther can'o'worms...'

Jayden933, 2021-11-03:
That....is very strange. I'm ssh'd into 129 and it's definitely 192.168.1.129 that my browser is currently pointing to for the UI. But changing it to 128, it pulls up the motioneyeos login and I used my credentials I set up on 129 and sure enough, it pulled up the exact same camera view with the same settings and everything. So it's...on both?

I ssh'd back into the PiZero at 130, and when I ping 192.168.1.128, I get nothing, same as before with 129. And when I tried traceroute 192.168.1.128, it went the full 30 hops and didn't report anything. Just * * *.

When I was first setting it up, I had to reimage the Pi3 one additional time for some reason, but don't remember why. Maybe the second time it set itself up, it got assigned 129 but internally kept 128 too or something? 192.168.1.129 is the address I set up port forwarding for on my router to access remotely and that's still working fine. I never used a static IP file on the Pi3 installation though.

Would it be worthwhile just to reflash both of them and set them up on new static IPs?

starbasessd, 2021-11-03:
In my opinion? Yes. 
I would have the router issue the static IP address to their MAC addresses. Leave the network settings at DHCP on both Pis. The only other thing I would check would be if you are accessing from a PC on the home network, what are its IP, GW, etc... It's one of those errors in setup where you'll hit yourself in the forehead and go ...

Jayden933, 2021-11-04:
I'm at a loss. I'm trying to start from scratch, cleared the devices from my router and everything. I wiped and reimaged the dev image for the PiZero, put just the wpa_supplicant in at first. Plugged it in, it connected to the internet because it got the time, the router sees it at its usual 192.168.1.130, but when I go there in a browser, I get nothing. When I try to ping from my PC, I get "Destination host unreachable" again. So I thought, maybe it needed the static_ip.conf to tell it what IP and gateway to use, even though the ones it selected were good. Re-imaged with the dev build, put the wpa_supplicant and static_ip from before in, it connected again just fine, but I still am unable to reach it. This is the exact same configuration that worked earlier from swapping to the dev build, I don't know of anything that's changed.

starbasessd, 2021-11-04:
Do you have an IP scanner program? Are you running Windows or something else?

Jayden933, 2021-11-04:
Windows, and no but I can see the new device and its IP pop up on my router. But I plugged a keyboard into the PiZero and started pinging my desktop PC (the one that just couldn't reach it minutes ago) at 192.168.1.65 and it worked. Then I went to .130 in a browser on the PC and it pulled up, so I don't know if it needed to just wait a few minutes or what. The PiZero never restarted or anything. I'm gonna hope it keeps working and move on to the Pi3 now

starbasessd, 2021-11-04:
The question is:
Why is it getting 2 IP addresses? 
Or why does it respond to 2 IP addresses?
Can you 'temporarily' turn off / disable / remove the port forwarding on the router?
Run a clean network. Lets get it working locally for you before poking a hole in
the router firewall.

starbasessd, 2021-11-04:
A good IP Scanner program I use is:
https://www.advanced-ip-scanner.com/
Comes in handy for these kinds of situations.

Jayden933, 2021-11-04:
Right, yeah I removed the port forwarding I had on 129 and cleared the devices list so it appears that 128 and 129 are unallocated. PiZero is fine for now, just on the single .130 IP address. I'll try flashing the Pi3 without a static_ip.conf first and see if it connects via wifi and on what IP

starbasessd, 2021-11-04:
I thought the Pi3 uses ethernet?

Jayden933, 2021-11-04:
It does, but if I _can_ get it working via wifi, I think I may prefer that long term. Unless you think leaving it wired is best?

Jayden933, 2021-11-04:
could that be why it took 2 IPs before? Connected one on wifi and the other via ethernet?

starbasessd, 2021-11-04:
Yes, if you enable WiFi and have ethernet connected, it will grab 2 IP addresses. 
If you are doing static, motionEyeOS prefers ethernet over wifi, so depending on
if you use both the static_ip.conf file and try to give it a conflicting ip address to the
wrong mac address from the router, you get weird things happening.

starbasessd, 2021-11-04:
If you plan to add cameras or other Pis, you really want your Hub (the Pi3 currently) to be connected via ethernet. You can also overload your WiFi if you have too many devices connected that are 'high bandwidth' users...

Jayden933, 2021-11-04:
I think I will stick to ethernet for it then. That may be what got me into this mess to begin with. Plus if I ever connect more in the future, a hardwired connection to the base station would probably be the most stable. I'll leave out any wpa_supplicant or static_ip files from the Pi3's boot folder and connect it to ethernet and report back

Jayden933, 2021-11-04:
Okay, Pi3 booted up fine. It's on ethernet only at 192.168.1.128. I can pull it up in a browser fine. I had to unplug and replug the PiZero to move it, and while it's showing up on the router device list at 192.168.1.130, I can't open it in the browser or ping it from my PC just yet. But this is what it was doing before, so I'll give it a few more minutes.

Jayden933, 2021-11-04:
Still nothing.....I waited for a while, but still couldn't reach it at 192.168.1.130. The router still sees it and shows an active connection with it. I just unplugged it and plugged it back in again. Not sure what the deal is with it

Jayden933, 2021-11-04:
I don't understand why, but for some reason it seems that I have to plug a keyboard into the PiZero, sign into it, ping my home PC at 192.168.1.65, which works on its end without issue, then magically my PC can ping it and the web UI instantly responds
¯\_(ツ)_/¯

Jayden933, 2021-11-04:
Regardless, after a little configuration I think this will be setup exactly as I was wanting. Thank you IMMENSELY for all of your help @starbasessd ! I will be sure to post here if I run into similar troubles in the future

Jayden933, 2021-11-04:
I made the mistake of changing the timezone on the PiZero which required a reboot. And I did have to go through the whole rigmarole of logging into it and pinging my computer before I could access it again. I don't know whether that's a bug or an issue with my local network.

starbasessd, 2021-11-04:
If it were a bug, there would be MANY reports here. I am unable to reproduce your issue here. Most issues with motionEye or motionEyeOS I can reproduce here in my 'lab' environment. (Pi1 - Pi4B-4GB & -8GB, PiZeros, PiZeroWs. PC, VirtualBox Host for VMs, Gigabit internal network, 400Mb/s Internet connectivity, etc., etc., etc... Not to mention my 'production system' with 3 d-Link DCS-502-Ls, 2 Foscams, 1 Tiga generic dashcam. Additional testing cameras include PiZeroW PiCam, another Tiga, a DSLR, and whatever I can find to throw at it. (Couple of NASs, lots of USB dongles, etc.)

Jayden933, 2021-11-04:
Lol yeah if you have it working in your plethora of use cases, I'm sure it can work here! I imagine it's probably something to do with my access point, or something I'm doing wrong with configuration. But I'm satisfied with this setup for now. Thanks again!

Jayden933, 2021-11-04:
I sort of have an update. So I added the PiZero camera to the Pi3 hub's web interface and left it overnight. I can connect to the Pi3, which is connected to the network via ethernet, at 192.168.1.128 no problem. And the Pi3 still sees the PiZero's camera feed no problem. But my desktop can't open the web UI for the PiZero at 192.168.1.130 and I'm unable to ping it again. Could it be that my router isn't allowing wireless devices to connect to other devices on the network if there's not some kind of handshake or regular communication between them? That would seemingly explain why the Pi3 and PiZero can still see each other since they've been in constant communication and why I have to initiate a ping to my PC from the PiZero before the PC can see it.

starbasessd, 2021-11-04:
If that is the case, I may be able to suggest something:
ssh to the PiZero
run the command:
crontab -e
add 2 lines: 
5 * * * * /bin/ping -c 2 192.168.1.128 >> /data/log/ping.log
\@reboot  /bin/ping -c 2 192.168.1.128 >> /data/log/ping.log
then hit \[ctrl\] x, y, \[enter\] to save and reboot \[enter\] to reboot
This should send 2 pings every 5 minutes to your hub and at startup, hopefully alleviating the issue. If you look in the folder /data/log you should see the ping.log with the last time it tried to send the ping (timestamp of the file) and the results. I set it to 2 in case the first one fails. I set it to 5 minutes because I don't know how often it needs to check. Once you see it's working, feel free to change the time and delete the ">> /data/log/ping.log" or change it from ">>" to a single ">" to overwrite the log each time, keeping only the latest result.

Jayden933, 2021-11-05:
Oh brilliant! I'll give that a shot. If I want to have it ping my desktop PC as well so I can continue to access it from a web browser there, I assume I should also add two additional lines with the IP of the desktop?

Jayden933, 2021-11-05:
I guess specifically, would these lines be good? I would prefer it to just overwrite the log.
```
*/5 * * * * /bin/ping -c 2 192.168.1.128 > /data/log/ping.log
*/5 * * * * /bin/ping -c 2 192.168.1.65 > /data/log/ping.log
@reboot /bin/ping -c 2 192.168.1.128 > /data/log/ping.log
@reboot /bin/ping -c 2 192.168.1.65 > /data/log/ping.log
```
or do I need to combine those somehow?

starbasessd, 2021-11-05:
I would not do so. If it works with pinging only the desktop, then you could just change the IP to ping to the desktop IP. If you must ping both, then change the file name to pinghub.log for the  hub and pingdesk.log for the desktop. 
Remember, this is a work-around, not a fix, to make it so you don't have to connect a keyboard to the PiZero to get it to connect. Ideally, you want to find the root cause...

Jayden933, 2021-11-06:
Thanks for all your help and info. I did end up just using the lines with the desktop IP, since it seems to have no problem communicating with the other Pi. At first, it didn't really seem like it made a connection in the first few minutes after it rebooted. (It doesn't need any kind of info for the admin login, does it? I did add a password.) But I let it sit for a while after restarting and tried connecting a while later from my desktop and it worked, so that's good

starbasessd, 2021-11-06:
So, if the reboot isn't pinging as expected, you can mod the line to say bootping.log, and then see if the @reboot line is working or is it pinging before the network is fully up. If it isn't working right away, but the 5 minute ping IS working, then there's another minor tweak you can add. 
Change the @reboot line to be at the end  bootping.log
Let me know if it is successful or fails. If it's failing, I'll give you the tweak...

starbasessd, 2021-11-06:
As to needing a password, no. It's already running as root/admin. The worst case scenario is you boot up at 11:30:01 and have to wait until 11:35:00 before the */5 ping is triggered...

Jayden933, 2021-11-08:
Sorry, was away for a while. But I did end up rebooting the Pi Zero because I changed its location, and it did seem to quickly reconnect after powering on again. So I think it's working properly!

starbasessd, 2021-11-08:
Thanks, my suggested mods... Glad to be of help. Don't forget to close after a few days, unless it breaks again.

Jayden933, 2021-11-09:
So far, all is good! I unplugged and moved the PiZero to another location, plugged it in, and was able to pull the IP up with its web interface on my desktop PC in less than 30 seconds. I'm going to get an additional PiZero to try to add as another new camera on Thursday. I'll try it first without your crontab mod and see if that works, but it's good to know I've got a solid workaround if it's needed. I'll have to try and investigate further if there's some rhyme or reason to why wireless devices on my network seem finicky with what connects to them. Thanks again so much for all your help!!

closed 2021-11-09 by Jayden933