motioneye-project/motioneyeos #2832: dev20201026 is not allowing to change camera settings.

mk48, 2021-11-04:
### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry PI Board camera.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Not using any peripherals.

### Log Files

I consider the following log files relevant to this issue:

* [messages.log](https://github.com/ccrisan/motioneyeos/files/7475904/messages.log)
* [boot.log](https://github.com/ccrisan/motioneyeos/files/7475905/boot.log)
* [dmesg.log](https://github.com/ccrisan/motioneyeos/files/7475906/dmesg.log)
* [motion.log](https://github.com/ccrisan/motioneyeos/files/7475907/motion.log)
* [motioneye.log](https://github.com/ccrisan/motioneyeos/files/7475908/motioneye.log)

## My issue
motionEyeOS dev20201026 won't allow changing camera settings. The left side pane is greyed out.
Camera is connected and showing video, but the settings page is disabled.
[image]

### Error message in web console
```
main.js?v=0.42.1:2341 Uncaught TypeError: Cannot set properties of undefined (setting 'checked')
    at dict2CameraUi (main.js?v=0.42.1:2341)
    at main.js?v=0.42.1:3330
    at Object.onResponse [as success] (main.js?v=0.42.1:452)
    at c (jquery.min.js?v=0.42.1:3)
    at Object.fireWith [as resolveWith] (jquery.min.js?v=0.42.1:3)
    at k (jquery.min.js?v=0.42.1:5)
    at XMLHttpRequest.r (jquery.min.js?v=0.42.1:5)
```

starbasessd, 2021-11-04:
Usually indicates not logged in as admin, or the motioneye.conf file got corrupted.
If you changed the password, or the username and password for admin, it may also be that the 'Remember Me' box was checked, and you can click on the user (head) icon and log in using the new username and password.

mk48, 2021-11-08:
I am using older verion (motionEyeOS 20190911). I think we can close this issue for now. I will try the latest version after sometime.

closed 2021-11-08 by mk48