motioneye-project/motioneyeos #2833: Unable to open video device, usb camera intermittent issue

sts-ryan-holton, 2021-11-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

- dev20201026

### Board Model

- Raspberry PI 4B 4GB

### Camera

- I'm using a USB webcam, from the brand Benewy, [this one](https://www.amazon.co.uk/Microphone-Desktop-Conference-Studying-Compatible/dp/B089LTY8X7/ref=mp_s_a_1_1_sspa?dchild=1&keywords=benewy+webcam&qid=1626897356&sprefix=Benewy&sr=8-1-spons&psc=1&smid=A3J9ITXBRZ4BLV&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExUjExUFhSNUNSSENRJmVuY3J5cHRlZElkPUEwNzA1MTkwM0Y0SExDTUpSUTlKMSZlbmNyeXB0ZWRBZElkPUEwNTc5MTk3VVlFTEJIUUlaN1o1JndpZGdldE5hbWU9c3BfcGhvbmVfc2VhcmNoX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=)

### Network Connection

- Wi-Fi

### Peripherals

- Just a USB camera

### Problem

I'm experiencing an intermittent issue whereby the video feed would show a grey screen with an "unable to open video device", and it shows that in the video recordings that are generated and saved as well.

What's weird, it will then suddenly recover itself after a few hours, but it means that for potentially hours on end there's no video captured.

I've tried different USB ports and have tried restarting many times. I shall attach the errors from the messages log file since that file seems to hold the errors.

```
Nov  6 05:23:11 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
Nov  6 05:23:11 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Nov  6 05:23:20 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Nov  6 05:23:20 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:20 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:21 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:21 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:21 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
Nov  6 05:23:21 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Nov  6 05:23:30 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Nov  6 05:23:30 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:30 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:31 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:31 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:31 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
Nov  6 05:23:31 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Nov  6 05:23:33 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 45 C
Nov  6 05:23:40 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Nov  6 05:23:40 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:40 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:41 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Nov  6 05:23:41 meye-1c24c44b user.err motion: [1:ml1:Car Cam] [ERR] [VID] vid_start: V4L2 device failed to open
Nov  6 05:23:41 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
Nov  6 05:23:41 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Nov  6 05:23:50 meye-1c24c44b user.warn motion: [1:ml1:Car Cam] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Nov  6 05:24:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 45 C
Nov  6 05:25:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 46 C
Nov  6 05:26:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 46 C
Nov  6 05:27:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 45 C
Nov  6 05:28:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 46 C
Nov  6 05:29:34 meye-1c24c44b user.notice throttlewatch: currently: ok, temperature: 47 C
Nov  6 05:30:00 meye-1c24c44b cron.info crond[572]: USER root pid 15866 cmd /usr/sbin/dyndns-update
```

starbasessd, 2021-11-06:
Interesting.
Can you copy the oops.py file in the zip to your /boot partition (whether on a SDCard or USB drive (you don't need to re-image) on the machine you used to image the SDCard or USB drive, please? Just shut down the Pi, pull the SDCard or USB drive, re-insert to the machine you imaged it on, find the /boot partition (the first partition of the drive) and copt/paste the oops.py file there, safely eject the drive, re-insert to the Pi, and turn it back on.
When the camera is 'missing' again, ssh into the Pi and run /boot/oops.py and report the result, please.
I know it's not thermal throttling (Temps in the mid 40's) but I want to check voltage throttling, due to the USB Camera dropping out.  
You can also check the camera in Raspberry Pi OS by running something like 'Cheese' or VLC or Windows (VLC) just run continuously for a while (as long as you were running in MotionEyeOS) to see if it drops out there, too.

starbasessd, 2021-11-06:
Here's the zipped file oops.py, it checks throttling and tells you what caused it.
(You can also check via:
vcgencmd get_throttled
but it just gives a number, but oops.py decodes the number to a list of issues.

[oops.zip](https://github.com/ccrisan/motioneyeos/files/7490365/oops.zip)

TheOtherDavidCook, 2021-12-29:
I have the same issue... and MotionEye continues to select the INCORRECT VIDEO DEVICE...

It assigns /dev/video10, which is wrong.. the cam works fine when VLC opens /dev/video0

I changed the device in /etc/motioneye/Camera1 but the config doesn't change in the browser

Apparently this setting is referenced somewhere besides /etc/motioneye/Camera1... and /var/lib/Camera1 is empty...

Who's gonna be nice and point me to the right file?

starbasessd, 2021-12-29:
I, too have a face for radio...

starbasessd, 2022-02-26:
@rpforbes Please open a new issue, so you can answer the standard questions. You didn't state which version of motionEyeOS, or which Pi, or anything else that would be pertinent.

closed 2022-02-26 by starbasessd