motioneye-project/motioneyeos #2834: Support for the new (official) PoE hat

Femah, 2021-11-06:
This is a question based on an earlier issue #1636 (support for the official PoE hat).

It seems that the Raspberry Pi Organisation has released a newer version of the PoE hat that is now incompatible with MotionEye.

I am running MotionEye 20201026 on a Pi 4. The fan on the PoE hat does not start despite the unit getting very hot. I have done a reasonable amount of reading on this, and so I downloaded and installed the latest version of Raspberry Pi OS. The fan on the heatsink did not start until the OS was updated. When I'm running Raspberry Pi OS, the fan works as expected. When I swap the SD card back to MotionEye, the fan does not spin.

When I read in issue #1636 that the original PoE hat had the same issues until you made some changes, it made me wonder whether anyone else is having the same issue and whether help might be on the way.

Thanks in advance.

starbasessd, 2021-11-06:
If you can figure out how it's triggered (see https://github.com/ccrisan/motioneyeos/issues/1636#issuecomment-424536198) If it's a simple dtbo file from /boot/overlays on a RaspberryPiOS you could copy that (those) files into the boot partition right after writing the image, and edit the cmdline.txt and/or config.txt files as needed. The older PoE hat seems to use /boot/overlays/rpi-poe.dtbo. 
Actually, looking at a Pi4 with a fully updated RaspberryPiOS the file you probably need (to start) is rpi-poe-plus.dtbo
How it's used I haven't researched (yet) I don't have a PoE+ to test...

Femah, 2021-11-07:
Thanks for the speedy response. I tried copying the overlay file from the "working" SD card to the MotionEye card - no change. I tried various changes to the config file - no change. I re-flashed the MotionEye card and copied the file before the initial boot - no change. I gave up at this stage and moved on to Plan B. My starting point is the latest Raspberry Pi OS Lite, all patches so the PoE fan works. Then I followed the MotionEye installation instructions. I am running MotionEye on a Pi with a PoE2 and working fan, but...the stream is not showing up on port 8081. I have access to a cut-down UI on port 8765 (no expert settings or network config). I'm almost there, but really need to know where I can access the stream.

starbasessd, 2021-11-07:
I don't have one to play with here, so I can only point you to the equivalent to the the #1636 comment... There may be apps, settings in /boot/config.txt that need to be added or changed, etc. the best option, maybe, would be to copy the cmdline.txt, config.txt and or figure out how / what was needed to get it active on the RaspberryPiOS...

starbasessd, 2021-11-07:
As to the motionEye issues, which instructions did you follow?
You cannot access the stream until the camera is added and set up in motionEye.
Did you use 'admin' and a blank password to connect to http://ip_address:8765?
Which browser are you using?

Femah, 2021-11-07:
> I don't have one to play with here, so I can only point you to the equivalent to the the #1636 comment... There may be apps, settings in /boot/config.txt that need to be added or changed, etc. the [...]

I made some progress. It seems that I am missing a virtual device - sys/devices/virtual/thermal/cooling_device0
I am no expert in this area, but I think the entries in the config.txt that access the rpi-poe-plus overlay are failing because the device can't be found.

After a frustrating day, I came up with Plan C. I am now using MotionEyeOS and I've hard-wired the fan to be permanently on (full speed). For the last hour, the reported temperature has been around 38 - 40C. I now have the full capabilities of the wonderful MotionEyeOS and PoE and cooling. (Response to your second comment is below).

Femah, 2021-11-07:
> As to the motionEye issues, which instructions did you follow? You cannot access the stream until the camera is added and set up in motionEye. Did you use 'admin' and a blank password to connect to http://ip_address:8765? Which browser are you using?

The instructions I followed are there ones: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
I installed the software and then physically connected and configured the camera. The camera image appeared fine through the regular UI (at the 8765 port), but the streaming port 8081 refused my connection attempts. I have several MotionEye installations, all using MotionEyeOS and working well, so I knew that there was something not right my installation, I just could not figure it out - (see above for Plan C).
I really appreciate your help. If you email me directly, we could discuss the feasibility of me sending you a PoE Plus hat to you could update MotionEyeOS. That way everyone wins :-)

starbasessd, 2021-11-07:
If I were a programmer, or had the ability to trigger a new build, I would be happy to help. I am but a Level 3 Enterprise Help Desk with delusions of Godhood. I do scripting, I do debugging, I do suggestions (but never for purchases of cameras, love, or horse racing). Thank you for the offer.

Femah, 2021-11-08:
Just in case anyone else finds this looking for answers. I have now managed to get Motion (and MotionEye) working on the latest RaspberryPi Lite (the one that correctly supports the PoE+ hat). The problem was that motion.conf was not being updated via the MotionEye UI. The motion.conf also has a setting to block streaming to computers other than "localhost". By default this setting blocks external requests to the streaming port (8081). By modifying motion.conf to point to the motioneye camera configuration files, and setting the streaming restriction "off", I now have what I need.

# Restrict stream connections to the localhost.
stream_localhost off

I hope this helps someone else.

starbasessd, 2021-11-08:
In motionEyeOS, it is already taken care of.

hcg1967, 2022-01-31:
Has there been any developments yet? I am facing the same problem as Femah of using the POE+ hat (on a RPi 3B+) and when using the latest MotioEyeOS image, the fan doesn't spin, no matter what I try.  RPi is running very hot, to the point where I am getting concerned it will be damaged permanently. Using installed MotionEye on the RPi OS, the fan works OK  but I experienced many problems in terms of stability. MotionEyeOS seems to work best for me but it's frying my Pi... Very frustrating indeed....

starbasessd, 2022-01-31:
See comment [here:](https://github.com/ccrisan/motioneyeos/issues/2839)