motioneye-project/motioneyeos #2836: New Build Request

starbasessd, 2021-11-08:
Preliminary Docs

I confirm that I have read the CONTRIBUTING guide before opening this issue.

I confirm that I have read the FAQ before opening this issue. YES
motionEyeOS Version

I am running motionEyeOS version: Current/Latest 20200606 & dev20201026
Board Model

I am using the following board/model: All supported Raspberry Pi boards (maybe others).
Camera

I am using the following type of camera: Any Supported.

My camera model is: Any Supported.
Network Connection

My motionEyeOS unit is connected to the network via: Any Supported.
Peripherals

I am using the following peripherals that I consider relevant to this issue: Any Supported.
Log Files

I consider the following log files relevant to this issue:

    (attach e.g. motioneye.log)
    (attach e.g. boot.log)

RaspberryPiOS has (literally 30 Oct 2021/ 8 Nov 2021) just upgraded to Bullseye (Debian 11 derivative). I t would be a Good Thing(tm) to run a new build, considering the age of the last build, the new boards available in the Pi line (Pi4-4GB, Pi4-8GB, PiZeroW2) that are all supported by the last Buster build and the new Bullseye build, but not fully supported by even dev20201026.
Thank you for your consideration.

wpa-3, 2021-11-13:
Was just about to make a build but might hold on a little bit, This would be great for the new pi02 and pi4 4gb/8gb

swolekat, 2021-11-14:
Looking forward to this. Currently I have to run the dev build to get my raspberry pi 4. I also think the performance isn't what it should be: I can currently only run 2 webcams at 240p at 10fps.

starbasessd, 2021-11-14:
I run 6 cams, 4@ 640x480 (mjpeg), 2@ 1280x720 (rtsp) at an average of 10FPS both on a Pi4-4GB (motionEyeOS dev20201026) and production on an I5 M73 Lenovo TFF PC running Ubuntu 20.04 LTS Server & motionEye.