motioneye-project/motioneyeos #2837: The Schedules is not working

peterd550, 2021-11-08:
I have set the schedule so motion eye to record from 7 am 10 pm but it has still recorded video

[image]

starbasessd, 2021-11-08:
Is the Time Zone set correctly and the time & name overlays correct?

peterd550, 2021-11-08:
Yes that's the first thing I thought of.
All correct and set to Melbourne and is overlays are showing correct time

I am using this as my sntp server  1.au.pool.ntp.org

peterd550, 2021-11-08:
I have SSH into the Pi and it is correct

starbasessd, 2021-11-08:
I'm sorry... Which OS? 20200606 or dev20201026?

peterd550, 2021-11-08:
I am using dev20201026

starbasessd, 2021-11-08:
I'll check and let you know, I'm on another Bullseye issue...

peterd550, 2021-11-09:
Any chance you had a look into this issue?

starbasessd, 2021-11-09:
Not yet, I was working on a Bullseye issue. #1 on my hit list for today after a nap...

peterd550, 2021-11-09:
Ok thanks 
I'm AEST so bedtime tomorrow

peterd550, 2021-11-09:
[image]

peterd550, 2021-11-09:
Here is a resent screenshot

starbasessd, 2021-11-09:
OK, dug a little deeper. My  mistake, I never noticed that Working Hours only applies to Motion Detection, not to all recordings. Thinking about another 'work around' for your circumstance/use case. Let me test a bit...

starbasessd, 2021-11-10:
Looking for a more elegant process than the 2 copies of the config file and switching them at morning and evening as posted before. It definitely could be an 'enhancement', since if you are not using motion detection, there is no way to use working times...

starbasessd, 2021-11-10:
Using crontab to simply start and stop recording using motion web controls does not work well due to too many checks needed to handle booting up / rebooting / power off / on cycles. Deep in thought. Even a 'simple' script is now over 30 lines... Depressing... I am still looking at it, though.

starbasessd, 2021-11-12:
Unfortunately, my skills are lacking. Bullseye is becoming like some M$ upgrades. Making changes, leaving stuff out, etc, just because they can. My brain is mush. 
I wrote a nice set of instructions at: https://github.com/ccrisan/motioneyeos/issues/2779#issuecomment-882773284 to resolve an issue where someone wanted to change a particular camera setting for overnight.
For you, the easiest would be to have your overnight settings be no Movie, and daytime enable Movie with Continuous Recording on. Unfortunately, Working Hours is only available (currently) with Motion Detection. I cannot figure out how to make it all work for your scenario / use case. 
I will leave this open as a Feature Request / Enhancement, to see if anyone else wants to take a stab at it, and maybe a Dev can come up with a better idea...

peterd550, 2022-03-27:
Used out of these hours and works well

closed 2022-03-27 by peterd550