motioneye-project/motioneyeos #2840: Raspberry Pi Night Vision Camera has IR on all the time

sobo82, 2021-11-18:
I got my Raspberry Pi Night Vision Camera which has IR and Night Vision view on all the time. Only when there is a lot of bright light (like an actual light shining on the small detectors next to the IR), the Night Vision turns off (tho keeps the IR on but thats no problem for me).
My plan is to put the camera in a bird nest cage so i bet it'll still detect it as too dark like 24/7

I cant find a setting inside MotionEyeOS to influence how bright or dark it has to be to trigger the night vision to be used. Am i overlooking things or do i need to work with a script?

If a script is the way to go, can someone point me out to something where it states that step by step on how to do such a thing. I have looked and googled many things but i cant seem to find anthing that helps.

**Details**
Hardware: Raspberry Pi Zero 2
Software: 20211113 release by jawsper
Camera: https://www.waveshare.com/RPi-Camera-F.htm 
Detected camera details:
  Camera Device vc.ril.camera 
  Camera Type MMAL Camera

CyanAutomation, 2021-11-18:
Not that it solves your direct issue, but thought I'd share a slightly different route that I took - which might be of interest.

I bought one of [these cameras](https://thepihut.com/products/raspberry-pi-night-vision-camera-ir-cut?variant=20063164268606), which has its IR LEDs and IR filter controlled by a 3.3v input into a camera control pin. I have a Wemos D1 Mini beside it, that flicks a GPIO to 3.3v at sunset, and then off at sunrise. If someone was so inclined, they wouldn't even need the D1 Mini, and just code a GPIO script directly on the Pi. A little bit of tinkering, but now good quality shots in both the day and night.

starbasessd, 2021-11-18:
@sobo82 There isn't a setting in motionEyeOS (or motionEye) to control those kinds of settings, as they are controlled by a sensor in most cameras. (and yours). If you can find a script to switch the camera from day to night, we can help you write one for motionEyeOS (or motionEye) like CyanAutomation suggests based on a number of different triggers...