motioneye-project/motioneyeos #2842: Question/Support - use as a "window"?

dvpierce, 2021-11-19:
Hi! I have dev20201026 running on a RasPi4 2GB. I have it in FNC mode and am using a 7" HDMI touchscreen as a live preview device.

I am wondering if there is a way to power the screen on/off, preferably with a tap on the touchscreen? (I understand that FNC mode precludes using the Pi for motion detection otherwise that would be perfect.)

starbasessd, 2021-11-19:
Are you using FNC HDMI Preview? As far as I know, no, you can't toggle the display on and off. Are you feeding the camera to another device, or otherwise need it on all the time, just toggling the screen? If not, you can just shut it down. 
I had a Kiosk set up on a stand alone Pi w/Official 7" display, rotating between 6 cameras. Unfortunately, it required RPiOS to do it, with motionEye...

dvpierce, 2021-11-20:
Yes, I am using FNC HDMI Preview. Ok, I will look into other options. Thanks!

starbasessd, 2021-12-30:
No further after 4 weeks

closed 2021-12-30 by starbasessd