motioneye-project/motioneyeos #2843: Lack of admin password leaves many publicly availble system vulnerable to RCE

pizza-power, 2021-11-21:
As with many web-based apps, the admin user can run arbitrary code on the underlying system via the web-gui. While that may or may not be a security issue, and for meye use cases it probably isn’t, I feel that the lack of password on the admin account has left many publicly available systems open to exploitation. 

I’d suggest a feature to implement a random password on initial installation. Off hand, I don’t really know how that would look. I may be able to implement something if I find time. 

Here is a link to my post about the code execution: 

https://www.pizzapower.me/2021/10/09/self-hosted-security-part-1-motioneye/

Long story short - an admin can run arbitrary code via the option to run commands when motion is detected (obviously). Additionally, an admin can upload a backup that contains a malicious tasks.pickle file that will execute arbitrary code.

starbasessd, 2021-11-21:
You have the ability to set the admin (root) password and a user password in the WebGUI. You can change their names too (but that may be problematic). They are set blank by default for ease of setup.

pizza-power, 2021-11-21:
> You have the ability to set the admin (root) password and a user password in the WebGUI. You can change their names too (but that may be problematic). They are set blank by default for ease of setup.

Yeah, I'm familiar with that. I'm just saying that a feature that automatically sets a random password on at least the admin account would be nice. As it stands, there are a lot of unprotected, internet facing installations out there. And they are ripe for becoming unwilling members of a botnet, or worse. 

Even a note on the installation page to suggest to users to set a password could be helpful.

starbasessd, 2021-11-21:
Random passwords are easily forgotten, especially if YOU didn't create it.

scerazy, 2021-12-19:
Do not leave empty password & do not make the system open to Internet! Common sense

closed 2021-12-19 by starbasessd