motioneye-project/motioneyeos #2844: Onvif Camera Triggered Motion Detection

jrubbel, 2021-11-24:
Is it possible to leverage the motion detection capability built-in into onvif cameras and subscribe to their events instead of linux machine performing motion detection to save cpu usage? onvif triggered motion detection would then just start the recording instead of analyzing the video stream for motion. Thanks

starbasessd, 2021-11-24:
It would take quite a few CPU cycles to monitor for the alert(s) from the camera(s).
Does your camera have the ability to record it's own video to a network drive?
{putting on my Kris Kringle hat from Miracle on 34th Street movie|
If it does, you could use your Pi as a NAS, instead of using motionEye or motionEyeOS, letting your camera do the heavy lifting. Your Use case points more to this solution than mE or mEOS...

jrubbel, 2021-12-07:
@starbasessd thanks for the response - it looks like there is some code on GitHub https://github.com/rostislavzz/motion-onvif-events - however, having switched from motioneye to Zoneminder, its not related to this community anymore. Thanks anyway!

closed 2021-12-07 by starbasessd