motioneye-project/motioneyeos #2845: First Bootup not going well, a lot of Errors

Daniel-Muszkiet, 2021-11-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20200606

### Board Model

Raspberry Pi Zero W

### Camera

5MP Camera for Raspberry Pi

I installed the motioneyesod 20200606 via balenaetcher and on the forst boot up this happens
[image]

and then this https://youtu.be/tmIROQIxrGQ

I have no Idea whats wrong. I tried the newer Version but it is the same issue.
Then i thougt maybe the sd card is broken but i tried it with my pi zero 2 and Octoprint and that works.
Then i wanted to test motioneyeos with my new pi zero and the errors above occure.
Can somone help please :)

starbasessd, 2021-11-25:
Which Pi? PiZeroW or PiZeroW2?
You should be using [this](https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) for PiZeroW
and [this](https://github.com/jawsper/motioneyeos/releases/tag/20211113) for PiZeroW2
I do not use balena etcher for imaging, I prefer [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/).
It's weird that it gets through the majority of the boot, then everything panics. It connected to wifi and gets the date and time, though. Almost everything in the YT video shows IO error panics, like it can't read or write to the SDCard. Have you tried another SDCard?

Daniel-Muszkiet, 2021-11-25:
Its the PiZeroW 
I ordered a new SD Card and try it tomorrow.

But the SD Card is working fine with OctoPrint installed on my Raspberry Pi W2

starbasessd, 2021-11-25:
Can you run RPiOS on that SDCard and the PiZeroW?

Daniel-Muszkiet, 2021-11-25:
I finally got it to work! :)

The problem was the sd card. I got one from a college and tested it and now it works great!
Did not think that its a sd card prolbem. On my GoPro the sd card works fine...

Well thank you all !

closed 2021-11-25 by Daniel-Muszkiet