motioneye-project/motioneyeos #2846: email notification on Motioneyeos PI ZERO 2

pdesimonee, 2021-11-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-11-29:
You need to answer the questions above, plus the additional ones I posted to you in the other issue.

pdesimonee, 2021-11-29:
I have installed MotioneyeOS on a new Raspberry PI ZERO2. At first did not work but now it works. I have also activated motion detection with email notification. The server runs on a raspberry PI3. The PIZERO is a remote motioneye camera. Also the Rasberry3 has a camera directly attached. So both are sending emails on motion detection. The mails from the RP3 have the attached pictures while the ones from the PIZERO have the attachments but they are empty. Identical configuration; identical recipient. Both to port 587, TLS.
motion gap 10
captured before 1
captured after 1
minimum motion frame 20

As I said the configuration is exactly the same on the PI3 and on the zero but the attachments from the zero are there but empty.

Thank you

starbasessd, 2021-11-29:
You must be using the image supplied by one of the contributors: @jawsper. I've tagged him to see if he has any ideas, as I don't have a PiZeroW2 to test. Do the images exist in the folder tree on the PiZeroW2?

pdesimonee, 2021-11-30:
yes, I followed the suggestion to download this "special" image. Obviously I am also waiting for the "official" one.

JS1Dev, 2024-01-29:
Hi @pdesimonee & @starbasessd, what's the best approach to get @jawsper 's Pi Zero 2 & MotionEye fix (see [here](https://github.com/ccrisan/motioneyeos/pull/2829/files)) to master so all updates will properly include it?

Context:
- At a glance, (and per @jawsper 's comment [here](https://github.com/motioneye-project/motioneyeos/issues/2827#issuecomment-1011372969)), the changes seem pretty small and low risk to merge (eventhough it's been 2 years since he made them in Jan 12, 2022) - imho [Pi Zero 2 W compatibility?](https://github.com/motioneye-project/motioneyeos/issues/2827#top) shoudn't have been closed until this was merged
- I bring this up as I'm also interested in Pi Zero 2 & MotionEye fix and figure others may have also run into this issue as the Pi Zero 2 was released on Oct 21 and others may be running into this as well

starbasessd, 2024-01-29:
@JS1Dev Sorry I cannot be of help, you would need to contact the devs that 'own' motioneye/motioneyeOS for the past 2 years. They said that motioneyeOS was not going to be a priority for them (mostly because they were not familiar with buildroot or ThingOS, IIRC). Maybe you could check with them to see if they wanted to have a new dev take it over, or maybe just simply fork it. There were some minor issues between the PI4 versions and the PI4 20200606, and sometimes even dev20201026 with the boot firmware and flash. And now there it the PI5 with PCIe capability... Or update the images for the previously supported SBCs and the latest kernels?

JS1Dev, 2024-01-30:
Hi @starbasessd, thanks so much for your response, I really appreciate it!  I'll do my best to contact the devs that 'own'  motioneye/motioneyeOS for the past 2 years as you suggested as you also mentioned on the [motionEye Google Group Forum](https://groups.google.com/forum/#!forum/motioneye) that they don't really look there.

pdesimonee, 2024-01-30:
I understand motionEye is dead

starbasessd, 2024-01-30:
@pdesimonee Dead? I don't know. Agonal breathing? Yep. I am here to give support to any users that are trying to use it, should they need it, because with the exception of the Python2 needed pieces, it still works and does the job, and is still better than most, if not all, the 'free' NVR systems out there. And I keep trying to get users with issues related to things I cannot help with to contact the current 'owner-devs' to get them interested in getting back to being involved, or, at the alternative, offer control to new devs like Calin did.

JS1Dev, 2024-01-31:
I've reached out to a couple of the devs yesterday and a couple today that I hope to hear back from.  @starbasessd, I hope you don't mind that I mentioned you in my request.

starbasessd, 2024-01-31:
I don't mind.