motioneye-project/motioneyeos #2847: Can you stream with the IP camera RTSP H264?

MinBYY, 2021-12-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera,

My camera model is : IPC-HFW1020S

### Network Connection

My motionEyeOS unit is connected to the network via : Ethernet

Can I get an IP camera RTSP H264 video?
Streaming in MotionEyeos is known as the MJPEG format.
Camera I use is an IP camera and is output to RTSP H264.
Is it possible to stream with an RTSP H264 image using an IP camera?
MJPEG occupies so much bandwidth that the frame comes out low.

Please help me.

starbasessd, 2021-12-02:
RTSP works best with dev20201026 or 20200606 found here:
https://github.com/ccrisan/motioneyeos/releases
iSpyConnect doesn't show that model being connectable:
https://www.ispyconnect.com/camera/ipc
I use 2 different Foscam using RTSP, without issues.

MinBYY, 2021-12-03:
> RTSP는 여기에서 찾을 수 있는 dev20201026 또는 20200606에서 가장 잘 작동합니다: https://github.com/ccrisan/motioneyeos/releases iSpyConnect는 연결 가능한 모델을 표시하지 않습니다: https://www.ispyconnect.com/camera/ipc 저는 2개의 다른 것을 사용합니다 문제 없이 RTSP를 사용하는 Foscam.

Thank you for your answer.
Do you think that if you connect it with RTSP, the video from the camera will be streamed in H.264 format? This is the most important thing for me.

I'm using an LTE router and the bandwidth is about 10Mbps.
I want to access Motioneyos from the outside and watch streaming videos.
If you test it with a USB camera that comes out as JPEG, it will be streamed about 3-4 frames.

The bandwidth is very limited, so we will use an IP camera that is output to H.264.
Do you think I'm right?

starbasessd, 2021-12-03:
I do not know. Your camera isn't listed in iSpy. I don't have one here to test. 
I do not know if it is truly h.264 or not. If you can connect to it using VLC, 
VLC can tell you what protocol the camera is using, and bitrate used.
Maybe someone else can help. Have you tried contacting the maker of the camera?

starbasessd, 2021-12-30:
No response 4 weeks

closed 2021-12-30 by starbasessd