motioneye-project/motioneyeos #2848: Find MAC Address

ryantechdev, 2021-12-17:
I'm looking for a way to find the MAC address of the Pi used for the cameras. The camera is at my non-tech-savvy grandparent's house 3+ hours away so I can't really ask them or just go there lol

No big deal if there's not a solution, just curious. 
--------
### Preliminary Docs

- [x] I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

- [x] I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

starbasessd, 2021-12-17:
Several ways, all require access to the network where it's connected

> Run command 'ip a' on the Pi in an SSH session or console
> look on the router's dhcp server leases
> In Win10 or Linux, run command 'arp -a' or 'arp -a [ip_address]

starbasessd, 2021-12-30:
Was any of this helpful?

closed 2022-01-07 by ryantechdev