motioneye-project/motioneyeos #2849: H265 stream

scerazy, 2021-12-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model

I am using the following board/model: Odroid HC2

### Camera

I am using the following type of camera: IPCam

My camera model is: Jennov

Kernel Version
Linux 4.9.84 #33 PREEMPT Fri Nov 8 12:37:54 CST 2019 armv7l
File System Version
MCE25_V0_BU_STARIR-TAOS V3.0.1.12 build 2019-11-28 15:31:49 

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

If I set camera on H265/H265+ then picture in VLC is solid, but MotionEYE gets image broken every few seconds.
Changing it to H264, picture stays solid.

So something is not right.

Not life threatening, just use H264

starbasessd, 2021-12-19:
Which image are you using?

starbasessd, 2021-12-19:
Which Jennov camera specifically?
The URLs to use vary greatly:
https://www.ispyconnect.com/camera/jennov
and can make a difference between h264 and h265 streams.

scerazy, 2021-12-19:
Who knows what model. It surely does not state in its web interface (and box is put away)

rtsp://192.168.100.9/stream0

starbasessd, 2021-12-19:
Really hard to attempt to help you if you can't help us help you.

scerazy, 2021-12-19:
What do you expect me to do? Invent model ?
It is not on the chassis, not on the non-existent box, not on ebay purchase receipt. Any more ideas?
In fact I do not need help, I just made an observation

closed 2021-12-19 by starbasessd