motioneye-project/motioneyeos #2850: 'Advanced Settings'

Willywireless, 2021-12-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314). 0.42.1

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+). PI 3 B+ OS version is Raspbian 11

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera). PI V2 1.8mp 1080 camera  which identifies as a MMAL  camera in the 'available list

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet). Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue: None

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I installed Motioneye following the Nov 17 instructions for on a fresh Bullseye installation which proceeded without any problems. The on board Pi camera as above identified as a MMAL camera and produces a stream as expected and I am able to connect it to my Blue Iris server.

However I have 2 questions:

On an earlier installation of Motioneye there was an 'Advanced Settings' option which allowed me to set the camera as a fast network camera using RTSP - but there is no 'Advanced Settings' page now??

I have changed backed to Motioneye having experienced some stability issues when previously using uv4l-raspicam. However although I appreciate that the PI camera does have relatively poor sensitivity to low light, with uv4l-raspicam I was able to set an automatic frame rate which produced much better night time images in the available ambient night light than I am now getting with Motioneye.  Is there anything I can do to the MotionEye configuration to get similarly improved low light images?

starbasessd, 2021-12-20:
There is a difference between motionEye and motionEyeOS. the 'Nov 17 instructions' were probably for motionEye on another OS, which doesn't support Fast Network Camera. If you need FNC or other motionEyeOS only features, you need Pi3B:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz

starbasessd, 2021-12-30:
Did this help?

starbasessd, 2022-01-18:
No reponse

closed 2022-01-18 by starbasessd