motioneye-project/motioneyeos #2851: Corrupted/noisy image before messing with streaming settings.

rzfzr, 2021-12-20:
Using a Raspberry Pi 3B+ and a logitech webcam, whenever the device boots up (like on a power outage) the camera boots up like this and all the recordings are corrupted:
[image]
If I change the streaming frame rate it fixes the image, and it records the files ok...
[image]

starbasessd, 2021-12-20:
Would you please answer all the questions presented when you opened the issue?

rzfzr, 2021-12-20:
Not sure how I missed them, sorry
I read the faq.
Running 20200606 latest
Pi 3B+
Camera is a logitech c310 via usb
Ethernet
Booted and recording to a 64gb sandisk sd card class 10 (real one)
This time it booted even worse, can't reconize anything:
[image]

[messages.log](https://github.com/ccrisan/motioneyeos/files/7747480/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/7747482/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/7747483/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/7747484/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7747485/motioneye.log)

starbasessd, 2021-12-20:
What happens when you run the dev20201026 image?
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz

rzfzr, 2021-12-20:
Just tried it, same stuff, do you think an older build would do it?
[image]

starbasessd, 2021-12-20:
Looking into more of the logs. There are numerous 
Jan  1 00:00:25 meye-5ad6fa34 user.notice throttlewatch: since boot: under-voltage throttled 
Jan  1 00:00:25 meye-5ad6fa34 user.notice throttlewatch: currently: under-voltage throttled , temperature: 32 C
The camera is pulling more power than the Pi or the power supply can provide. It does boot, but is hampered by the camera.

rzfzr, 2021-12-20:
I saw the lighting bolt on screen during the whole booting up process, and switched to a better powersupply, it now only flashes it once in a while, I'll buy something better yet and get back to you

rzfzr, 2021-12-20:
Or is there anything I could make to lower power consumption?

starbasessd, 2021-12-20:
You could use a self-powered USB hub.