motioneye-project/motioneyeos #2852: Wifi connection issues

lordmortis, 2021-12-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi W Zero

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry Pi Camera Board v2

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

None

## Notes
 
So before first boot I put this in the `wpa_supplicant.conf` 
```
country=AU
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    ssid="<SSID>"
    psk="<PASSWORD>"
}
```
where `<SSID>` contains alpha numeric and `-`, and `<PASSWORD>` contains alpha, numeric, `@` and `^` characters.

(I got this config fragment from [here](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration))

I get wifi on initial boot (and the camera and everything seems to be working), but when i reboot i get no wifi. Mounting the disk reveals this in `/etc/wpa_supplicant.conf`

```
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
}
```

Also, after rebooting the `wpa_supplicant.conf` file is missing from the boot partition.

writing a new `wpa_supplicant.conf` to the boot partition doesn't appear to change the situation. Also the second time i did it, the filesystem wasn't unmounted properly so the filesystem is corrupt.

lordmortis, 2021-12-25:
Oh also - the image was written via balenaetcher on macos. not sure if that would make a difference.

starbasessd, 2021-12-25:
Removal of the wpa_supplicant.conf file during/after first boot from the /boot partition is normal.
The /etc/wpa_supplicant.conf is normal.
The /data/etc/wpa_supplicant.conf is the active one. It should be a copy of the one you used in /boot

What, if anything, did you change during the first run? 

If you are trying to set a static IP, you may want to do it with 'static_ip.conf' placed in the same folder as wpa_supplicant.conf when you do the original image
https://github.com/ccrisan/thingos/wiki/static_ip.conf
I would also consider using the dev20201026 image, it is more stable.

lordmortis, 2021-12-27:
So the first time I didn't change anything, I just shut it down and moved locations and I had no wifi. The second time I changed one of the motion eye settings that required rebooting and the system never came back up. 

I'm not trying to set a static IP - i handle all that on my router end via static DHCP leases, I just need it to keep connecting to my IoT wifi network.

lordmortis, 2021-12-27:
I remember what I was changing - I changed the timezone and the system rebooted. I'm trying that again with the dev firmware you reccomended now

lordmortis, 2021-12-27:
That firmware does appear to work considerably better!

lordmortis, 2021-12-29:
Yep, given I've rebooted multiple times and everything seems to be fine now, I'd say it was the image i was using.

closed 2021-12-29 by lordmortis