motioneye-project/motioneyeos #2853: Auto email notification

myscho15, 2021-12-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: latest.

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

Rpi HQ

### Network Connection

Wifi

### Peripherals

I am using lipo bateries

Hallo all,

I am using my project as security camera, work very well, but i am using 4G router. Rpi and router is connected to Lipo baterries (now is Winter and cold and capatity of bateries is smaller) and sometimes i have problem with router. In motioneye OS i am using motion triggered photos and then email notification, but sometimes a few days motion eye detenct no motion and i dont know if is working or is problem with router or bateries.

Can anyone please send me here a guide how can i setup in motioneye OS automatic daily notification via email? (Notification give me info - system is working and is alive. (Simple email like test email notification) this function will handy and Nice.

Thank all ;-)

starbasessd, 2021-12-28:
If you can connect to the Pi, you can run the command 
vcgencmd get_throttled 
to see if it has run into low voltage issues.
Will take a look at your auto-email request shortly.

myscho15, 2021-12-28:
Hay thank for the answer.

i dont have problem with low voltage (i have 3s lipo with BMS and rpi is connected with micro usb (usb with voltage step down). In this case i cannot have problem with low voltage. But sometimes i dont have time get to my cabin and change bateries and i dont know if motioneye is running or not. Sometimes i get daily 100 email notification with 5-6images (deers, birds, sun etc) sometimes i get 5 notification daily, sometimes 0 (and that is problem, when is this day with 0 notification, 8day after change batery i dont know if camera still work or batery is death (depend on days before and how much e-mails send.  

Rpi i use for my project run 6month - i only change bateries- if is more days with many emails work 7-8days, but if is more days with few emails work 8-10 days.

i have sometimes problem with data on router and when i change batery and i forget check if i have enough data on SIM card then notification will not work (no data no e-mail:D). This is major problem i want to solve this problem with this notification. 

If i become every 12hour email with notification -i know camera is runing and my cabin is secure :) than will be nice.

starbasessd, 2021-12-28:
Never assume battery voltage is good, nor assume the Pi isn't overheating. Even on "Official" power supplies, when there is too much current draw, it can go 'low voltage', especially on batteries. The Messages log on 20200606 and dev20201026 show watchdog temperature and low voltage alarms. 
I am looking at something for your notifications.

myscho15, 2021-12-28:
Uff i didnt know that, i check that and send here the logs if is good or not. I use only small pasive heatsinks (lastly i check logs will temperature 48-57)- its Ok?. Battery is 54AH Sony vtc6. I use my device 6month and work very well. But this week i send here logs. Big thank you for you patiente and kindly answers.

starbasessd, 2021-12-28:
48-57 is OK. Still looking. Got to go prep dinner, so will be offline for a while.

myscho15, 2021-12-28:
Thank you dont rush that have time. I am going sleep -morning work. But i am very happy that good and helpfull people still live :-)

starbasessd, 2021-12-29:
OK. 
ssh into the pi with motionEyeOS
run the following commands
cd /data/usr
nano ./systemup.py
copy-paste the following:

>import smtplib, ssl
>port = 465 # For SSL
>smtpServer = "smtp.gmail.com"
>sender_email = "from_email@gmail.com" # Enter your gmail account address
>receiver_email = "to_email@gmail.com" # Enter receiver address (can be the same)
>password = "your_gmail_password"
>message = "motionEyeOS System is up."
>context = ssl.create_default_context()
>mailServer = smtplib.SMTP_SSL(smtpServer, port )
>mailServer.login(sender_email, password)
>mailServer.sendmail(sender_email, receiver_email, message)
>mailServer.quit()

Edit the marked lines to match your accounts and password
\<ctrl\>x y \<enter\> to save
Now enter
crontab -e
\* x,y \* \* \* python /data/usr/systemup.py  
\# x and y are hour of day you want notice e.g. 6,18 would be 6a and 6p
\<ctrl\>x y \<enter\> to save
You can test by running 
python /data/usr/systemup.py
and look for the email in your account.
Hope this helps, it's somewhat 'quick and dirty'...

starbasessd, 2021-12-29:
Any thoughts?

myscho15, 2022-01-11:
Hay, i dont test set in few days i will have device and i will try, your Command. Thank you i Wish will work. Sorry i have now Corona :-/ and feel bad.

starbasessd, 2022-01-11:
Hope you feel better soon.

myscho15, 2022-01-15:
Hay, first thank you for help. 

i make all you write and test email is working. i will wait to tommorow if will work auto.

I have learn something new : how to connect to device via ssh :D i am noob.

starbasessd, 2022-01-15:
I hope it works (does what you want / need) for you. I'm not a programmer, but I get along in python to a point. It is amazing what you can learn by accident trying to fix something else, like how SSH works.

myscho15, 2022-01-15:
Hay, i setup the time 10, 20 (10am and 8pm).

but 20:00 email sent (its work), but now come one email every minute. its something wrong :-/

myscho15, 2022-01-15:
> I hope it works (does what you want / need) for you. I'm not a programmer, but I get along in python to a point. It is amazing what you can learn by accident trying to fix something else, like how SSH works.

yeah you are programmer you create somthing with code... i am only user, i want know how to programming a try this few times, but i dont know ho start :(

starbasessd, 2022-01-15:
Post your crontab setting line for the command

starbasessd, 2022-01-15:
Here's a fair youtube video to help learn python:
https://www.youtube.com/watch?v=rfscVS0vtbw
There's a few good sites associated with RPiFoundation to learn the basics on their site:
https://www.raspberrypi.org/learn/

myscho15, 2022-01-15:
* 10,20 * * * python /data/usr/systemup.py

starbasessd, 2022-01-15:
Yes, that is the line. oops. Figured it out. Set the first \* to 0
That will cause it to only run at 10:00 and 20:00.
That makes the line:
> 0 10,20 * * * python /data/usr/systemup.py
My fault, my mistake...

myscho15, 2022-01-15:
Hay, thank you... what makes one letter :D i change to 21:00 i will wait for email.

Its nic know tha still exist people thay can and want help to another :-)

starbasessd, 2022-01-15:
I made my living as an Enterprise / Level 3 Help Desk person for the past 25+ years. It's what I do. I do this application (motionEye / motionEyeOS) as partial payback for an excellent system. I toss in things like this because I will use it in other systems I use...

starbasessd, 2022-01-15:
I don't know what time zone you are in, please let me know if that last change fixed your issue.

myscho15, 2022-01-15:
Its working, i will test it few days. I hope will work good. 

But i found a small problem: if both email is the same (sender and receiver) but i email message display only sender and  “Bcc” and dont display recierver adress.

if i setup sender email (one adress) reciever (second adress) display email on second adress but reciever in email message is missing.

[image]
[image]

its only small problem.

starbasessd, 2022-01-15:
iPhone?

starbasessd, 2022-01-15:
I only use GMail web browser mail not the android app. I don't know. I click the info arrow next to "Me" and it shows sender, who it was sent to, which email server(s) it went through, etc. Sorry.

myscho15, 2022-01-15:
Yes, but in classic email from motioneye (is normal display sender and reciever) - email notification motion triggered. 

but this email “its alive” looks like photos i send.

its only small issue :-) will good fix this for future if anyone will want implement into motioneyeOS UI :-)

myscho15, 2022-01-15:
this looks classic email from motioneyeOS if detect motion…

[image]

but i am very gratefull for you help. :-) thank you.