motioneye-project/motioneyeos #2854: [motionEyeOS vs MotionEye installed on Raspbian BullsEye] Is there a backup incompatibility?

rtek1000, 2021-12-30:
Hello,

I backed up in motionEyeOS, and then tried to restore in MotionEye installed on Raspbian BullsEye, but after restarting the connection was refused.

Looks like the /etc/motioneye/motioneye.conf file has changed.

motioneye.conf - Before Restore:

```
# path to the configuration directory (must be writable by motionEye)
conf_path /etc/motioneye

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

# default output path for media files (must be writable by motionEye)
media_path /var/lib/motioneye

# the log level (use quiet, error, warning, info or debug)
log_level info

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 8765

# path to the motion binary to use (automatically detected if commented)
#motion_binary /usr/bin/motion

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# whether to restart the motion daemon when an error occurs while communicating with it
motion_restart_on_errors false

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 43200

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root) 
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media

# path to the wpa_supplicant.conf file
# (enable this to configure wifi settings from the UI)
#wpa_supplicant_conf /etc/wpa_supplicant.conf

# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime

# enables shutdown and rebooting after changing system settings
# (such as wifi settings or time zone)
enable_reboot false

# timeout in seconds to use when talking to the SMTP server
smtp_timeout 60

# timeout in seconds to wait for media files list
list_media_timeout 120

# timeout in seconds to wait for media files list, when sending emails
list_media_timeout_email 10

# timeout in seconds to wait for zip file creation
zip_timeout 500

# timeout in seconds to wait for timelapse creation
timelapse_timeout 500

# enable adding and removing cameras from UI
add_remove_cameras true

# enables HTTP basic authentication scheme (in addition to, not instead of the signature mechanism)
http_basic_auth false

# overrides the hostname (useful if motionEye runs behind a reverse proxy)
# server_name motionEye
```

motioneye.conf - After Restore:

```
conf_path /data/etc
run_path /tmp
log_path /var/log
media_path /data/output
motion_binary /usr/bin/motion
log_level info
listen 0.0.0.0
port 80
mount_check_interval 300
motion_check_interval 10
cleanup_interval 3600
remote_request_timeout 10
mjpg_client_timeout 10
mjpg_client_idle_timeout 0
smb_shares true
smb_mount_root /data/media
wpa_supplicant_conf /data/etc/wpa_supplicant.conf
local_time_file /data/etc/localtime
enable_reboot true
enable_update true
smtp_timeout 60
zip_timeout 500
add-remove-cameras true
password_hook /usr/libexec/meyepasswd
```

[motioneye-config (1).tar.gz](https://github.com/ccrisan/motioneyeos/files/7791120/motioneye-config.1.tar.gz)

starbasessd, 2021-12-30:
Remember, there are several major differences between motionEyeOS and motionEye:
* motioneyeOS is a mostly ReadOnly file system, with 3 partitions: /boot / and /data, and it uses links and file system overlays to put writable folders and files in the /data partition
* motionEyeOS WebGUI is on port 80, motionEye is on port 8765
* "Settings", "Expert Settings" exists in motionEyeOS, but not motionEye
* Logs are different between the 2 systems
* Many functions in motionEyeOS (FTP and SAMBA services, mappings, etc) are dependent on services and clients installed in the base OS in motionEye. 
* Fast Network Camera only exists in motionEyeOS
motionEyeOS is a very stripped down system with just enough to enable motion, ffmpeg, and motionEye to run with very basic services. motionEye runs on an unlimited capability Linux OS...

cap9qd, 2021-12-30:
The main issue I had doing this (restoring a backup between a MotionEyeOS
install to a MotionEye on Ubuntu for an Odroid XU4/HC1/HC2 board) as all
the paths are different between a MEyeOS and MEye as @starbasessd mentioned
with the different partition structure of MEyeOS. Once I figured all that
out I was able to extract the backup with 7zip, fix the paths via NotePad++
to maintain EOL and re-TAR the files with 7zip I was able to get it to
work. It is almost less work to just use the backup files to rebuild
everything from the web GUI. But that is just my experience...

It might be easier to just fix the paths in the camera-*.conf file paths
and transfer them via WINSCP to the machine and add them to the motion.conf
file than to restore the backup tarball and try to fix everything.