motioneye-project/motioneyeos #2857: TP Link Tapo TC70 - grey box

atreeon, 2021-12-31:
I confirm that I have read - yes
I confirm that I have read the [FAQ] before opening this issue - yes
I am running motionEyeOS version: motionEyeOS 20200606
I am using the following board/model: Raspberry Pi 3 Model B V1.2
I am using the following type of camera: old usb logitech (shows up) & a Tapo IP TC70 (not working - grey box) 
My camera model is: as above
My motionEyeOS unit is connected to the network via: Ethernet
I am using the following peripherals that I consider relevant to this issue: none
I consider the following log files relevant to this issue: can't seem to see any log files but possibly I'm looking in the wrong place (let me know if you think relevant)
 
Hi, I've had motioneye setup for a while using a usb camera and it has worked really well however I've just received a Tapo TC70 IP camera and can't get it to show the image.  

I've set up my Tapo TC70 and I can access it using the Tapo software.  I have tried to connect it to my motioneyeos setup but I'm getting a grey box and the message 'unable to open video device'.  I used ispyconnect to get the protocol and path but unfortunately no image.  Any ideas on what to do?

It seems a bit like this issue - https://github.com/ccrisan/motioneyeos/issues/2826

starbasessd, 2021-12-31:
You tried all 3 URLs?
rtsp://ip_address/stream1
rtsp://ip_address/stream2
and rtsp://ip_address/
? There wasn't a resolution for the other issue, unfortunately, it 'just started working'.

starbasessd, 2021-12-31:
Can you post/upload the 5 logs, please? (zip them into a single file, please before uploading)

atreeon, 2021-12-31:
Thanks!

Yes, I have tried the all three URLs.

I've attached the log files
[Archive.zip](https://github.com/ccrisan/motioneyeos/files/7796729/Archive.zip)
e

atreeon, 2021-12-31:
This line looks relevant but it doesn't have a time attached to it

'netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera2): Server returned 401 Unauthorized (authorization failed)'

starbasessd, 2021-12-31:
Do you have the correct username & password entered into the Add Camera box for the camera?

atreeon, 2021-12-31:
Ah!  Yes, that was it.  Apologies, I was being dumb!  I had entered the motioneyeos username and password and didn't realise there was a way to set a username & password on the camera.  Thanks so much :)

starbasessd, 2021-12-31:
Looks like you also started to set up the GDrive, but never completed it correctly.
> 2021-02-23 19:27:54: [motioneye]    ERROR: gdrive: missing authorization key
> 2021-02-23 19:27:54: [motioneye]  WARNING: accessing gdrive failed: missing authorization key
If your camera IP_address is 192.168.0.12 more warnings that you don't have the correct username & password:
> 2021-02-23 19:51:30: [motioneye]  WARNING: 403 GET /login/?_=1614109890379&_username=_&_signature=c9f944a07158473309e5ed95ba6a33e6e5d51831 (192.168.0.12) 2.52ms

I would suggest starting over, because of many mis-setup issues. Try using dev20201026 instead of 20200606.

atreeon, 2021-12-31:
Thanks, I'll take a look at that

closed 2021-12-31 by atreeon