motioneye-project/motioneyeos #2858: Unable to open video device /grey output with otherwise verified working camera and compute module.

texxupport, 2022-01-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (dev20201026).

### Board Model

I am using the following board/model: (Picam module for compute module4 with a compute module 4 ) (also attempted to use official CM4 carrier board)

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (Raspberry HQ cam).

### Network Connection

My motionEyeOS unit is connected to the network via: (wifi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7799897/motioneye.log)

 * (attach e.g. boot.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/7799899/boot.log)

starbasessd, 2022-01-02:
Unfortunately, the motionEyeOS has not been tested with CM4, AFAIK. I would confirm it works with RaspberryPiOS (Buster or Bullseye) and motionEye. See instructions here:
https://github.com/ccrisan/motioneye/wiki/Installation  (in the right hand column)
There is unique hardware and requirements for CM4 boards that may not have been in the buildroot at the time of the last build, this would make sure that the motioneye and motion and ffmpeg works.

thartman83, 2022-05-23:
I have found a potential solution to this using the PiCam board and a CM4. Extract the dt-blob.bin file from the provided image for the carrier board found here: https://www.ledato.de/shop_content.php?coID=24&language=en
Drop the dt-blob.bin into the /boot directory and reboot. The camera should show up now as /dev/video0. I was able to add the camera in MotionEye as a local v4l2 Camera type using the mmal service 16.1 camera. Hope this helps.
[image]

starbasessd, 2022-05-23:
Thank you @thartman83. Hopefully the Devs pick this up when re-building mEOS.

VoeGalore, 2023-01-09:
> I have found a potential solution to this using the PiCam board and a CM4. Extract the dt-blob.bin file from the provided image for the carrier board found here: [...]

Were you able to use it as a drop-in in the motioneyeos img for RPi4? (say dev20201026 release) if so then did it work in fast camera mode? I'm only recently starts to test the picam CM4 board.