motioneye-project/motioneyeos #2859: ftp client saves as root on the remote filesystem

emsruderer, 2022-01-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-01-03:
Please answer the questions above.
FTP would save as whatever user you used to log into the remote system, as expected. 
What were you expecting?

emsruderer, 2022-01-03:
The credentials are from a commonuser, nevertheless the files are saved as root:root by the ftp client

starbasessd, 2022-01-03:
Again, it isn't the FTP client saving as root, it's how the FTP server is set up. Use another FTP client, on another computer, connect as that user, and I'll bet good money that they get saved as root:root.

closed 2022-01-04 by emsruderer

emsruderer, 2022-01-04:
I still have trouble with the ftp client connecting to a linux ftp server. Login succeeds but the connection times out. 
2022-01-04 13:57:08: [motioneye]    ERROR: ftp: [Errno 110] Connection timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 856, in test_access
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 927, in _make_dirs
  File "usr/lib/python2.7/ftplib.py", line 530, in nlst
  File "usr/lib/python2.7/ftplib.py", line 441, in retrlines
  File "usr/lib/python2.7/ftplib.py", line 378, in transfercmd
  File "usr/lib/python2.7/ftplib.py", line 337, in ntransfercmd
  File "usr/lib/python2.7/socket.py", line 575, in create_connection
error: [Errno 110] Connection timed out
2022-01-04 13:57:08: [motioneye]  WARNING: accessing ftp failed: [Errno 110] Connection timed out
2022-01-04 13:57:41: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2022-01-04 13:58:25: [motioneye]    ERROR: ftp: [Errno 110] Connection timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 856, in test_access
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 927, in _make_dirs
  File "usr/lib/python2.7/ftplib.py", line 530, in nlst
  File "usr/lib/python2.7/ftplib.py", line 441, in retrlines
  File "usr/lib/python2.7/ftplib.py", line 378, in transfercmd
  File "usr/lib/python2.7/ftplib.py", line 337, in ntransfercmd
  File "usr/lib/python2.7/socket.py", line 575, in create_connection
error: [Errno 110] Connection timed out
2022-01-04 13:58:25: [motioneye]  WARNING: accessing ftp failed: [Errno 110] Connection timed out
2022-01-04 14:00:55: [motioneye]    ERROR: failed to upload file "/data/data/home/ftp/2022-01-04/13-59-38.mp4" with service ftp: [Errno 110] Connection timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 876, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 927, in _make_dirs
  File "usr/lib/python2.7/ftplib.py", line 530, in nlst
  File "usr/lib/python2.7/ftplib.py", line 441, in retrlines
  File "usr/lib/python2.7/ftplib.py", line 378, in transfercmd
  File "usr/lib/python2.7/ftplib.py", line 337, in ntransfercmd
  File "usr/lib/python2.7/socket.py", line 575, in create_connection
error: [Errno 110] Connection timed out

ue Jan  4 13:45:58 2022 [pid 14267] CONNECT: Client "::ffff:192.168.20.45"
Tue Jan  4 13:45:58 2022 [pid 14266] [lookout] OK LOGIN: Client "::ffff:192.168.20.45"
Tue Jan  4 13:56:36 2022 [pid 16261] CONNECT: Client "::ffff:192.168.20.45"
Tue Jan  4 13:56:36 2022 [pid 16260] [admin] OK LOGIN: Client "::ffff:192.168.20.45"
Tue Jan  4 13:57:53 2022 [pid 16279] CONNECT: Client "::ffff:192.168.20.45"
Tue Jan  4 13:57:53 2022 [pid 16278] [admin] OK LOGIN: Client "::ffff:192.168.20.45"
Tue Jan  4 14:00:23 2022 [pid 16307] CONNECT: Client "::ffff:192.168.20.45"
Tue Jan  4 14:00:23 2022 [pid 16306] [admin] OK LOGIN: Client "::ffff:192.168.20.45"
Tue Jan  4 14:04:01 2022 [pid 16473] CONNECT: Client "::ffff:192.168.20.45"
Tue Jan  4 14:04:01 2022 [pid 16472] [admin] OK LOGIN: Client "::ffff:192.168.20.45"

starbasessd, 2022-01-04:
When you say "linux ftp server" what OS is it running on, and which ftp server is it? I'd like to try duplicating it.

starbasessd, 2022-01-04:
And you still didn't answer any of the above questions. WHICH motionEyeOS are you using? (Which install file are you using?) They will help us diagnose your issue.