motioneye-project/motioneyeos #2860: MotioneyeOS with support of minicom & HAT SIM7600

Tormund1985, 2022-01-05:
Hi to everyone,

I am working with Motioneye under Raspbian and MotioneyeOs with the IMX477 sensor and different lenses on a PI ZERO 2. I need the cameras to be connected to an LTE connection via the Waveshare SIM7600 HAT through the minicom and the Ngrok service.

I have achieved this under Raspbian, but however I would prefer to do it from MotioneyeOS itself as I find it faster and with better features like FastNetwork.

I've seen that it could be done by building a build of my own, but I'm think it could be too much hard for me.

Another option would be to use an external router and output it to the internet through a VPN or a service like Balena.

What would you recommend me to do? 

Thanks in advance, and thanks for such an interesting project.

starbasessd, 2022-01-05:
If there are special drivers needed for your hat, I'd try to figure out how to add them. If you are going to use a PiZero2W you'd have to use @jawsper image. Short of that, Build Your Own is the best option. Fast Network Camera streaming can be added to a motionEye (sorta) install using https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera streamEye (Buster, not Buster Legacy or Bullseye due to major changes they made to camera support and python modules, but that may negate using the PiZero2W for lack of support for it)... 
You are using a very unique setup...

starbasessd, 2022-01-05:
Let us know how it goes...

Tormund1985, 2022-01-10:
I am trying to set up the OS, but I am not able to see the option to include NGrok, vlc-server, dwagent and minicom to have 4G connection.... 

Another option would be to let me the system via terminal commands such as "apt-get install". 

Can someone help me to compile a suitable OS for my needs? 

Thanks.

starbasessd, 2022-01-10:
You are changing motionEyeOS from an intentional minimalist local OS to something else. This might be accomplished by finding a programmer to fork and make your changes. Buildroot might have the options to add apt functionality, but the moment you do so, you are switching over to RaspberryPiOS and might as well switch to RaspberrPiOS and motionEye using the instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
or
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye

Tormund1985, 2022-01-10:
I currently have a raspberry pi 4 8Gb and a pi zero 2 with raspbian working "fine" with motioneye. However I tested a pi zero 2 with motioneyeOS and found that it works much better with this minimalistic software. That is why I would like to modify motioneyeOS as I need the services I mentioned.... 

Maybe adding apt will help however if the system is in read mode it will be impossible to install anything.

starbasessd, 2022-01-10:
Which is why you need a complete re-compile, but to add apt to install all the other things needed to run the apps you want would effectively change it from motionEye super lite to RaspberryPiOS Lite (at best). You can disable Read-Only for the 2 OS partitions, but that negates the protection of having a Read-Only OS. In Other Words, In My Opinion, you might as well as go with RPiOS Lite and build your own from there. Or read the manuals for setting up your own Buildroot system including the motionEye system & full OS options, or, maybe using RPiOS-Lite, Docker & the motionEye Docker image. 
But if you insist on changing a Buildroot system to a full system you may want to try to get support over at:
https://buildroot.org/downloads/manual/manual.pdf
and
https://buildroot.org/support.html

Tormund1985, 2022-01-10:
Then I will try to install over the lite version because a complete rebuild it would be too much for me. 
Many thanks for your support, I really appreciate it.