motioneye-project/motioneyeos #2861: ssh configuration of motioneye units (no web access)

joanila, 2022-01-05:
I have 2 motioneye units (Raspi Zero W V1.1 with Raspi cameras) running as swiss watches for more than a year on a remote local network (in my second home, 600 kms away from where I live right now). I cannot access these 2 Raspis directly, but I have ssh access to a third Raspi on their same local network through a reverse ssh tunnel. Once ssh-connected to this Raspi I can ssh to the 2 other Raspis running motioneye os. In other words, I can ssh to the motioneye Raspis, but I cannot access them through the web interface.

Since I cannot access the web UI of the 2 Motioneye Raspis and I need to change their configuration, my question is how can I do what it can be done in the web UI (changing motioneye parameters, updating the software, etc...) through ssh.

Is this possible ? If so, is there some place where it is explained how to do it ?

Thanks very much for your help

Joan

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-01-05:
A couple of different ways you might be able to access the MotionEyeOS Pis:
> open a firewall port forwarding to each of the Pis, Outside port 9081 to Pi1 port 80, Port 9082 to Pi2 port 80.

> If you are running a desktop on the reverse ssh Pi, you could port forward the x session and run the WebGUI on it.

> Depending on what needs to be modified, you might be able to find the parameters in /data/etc/motion.conf, /data/etc/motioneye.conf or /data/etc/camera-1.conf and change them manually, then restart motioneye with the command   "service motioneye  restart"

Other options too. If you need more specifics on an option, let me know and I'll try to post for that specific.

joanila, 2022-01-05:
1- I cannot open any port on the router of the remote network because it is a CGNAT 4G router. In other words I can exit the network, but I cannot enter on it with NAT (port forwarding is on the hands of the ISP). This is why I'm running the reverse tunnel.
2- Forwarding the x session through the existing reverse tunnel probably is an option but right now I don't know how to do it (I learned a lot but I spent quite a lot of time putting to work the reverse ssh tunnel). I must explore this option.
3- One of the things that I would like to do is to rotate the images and update motioneyeos. Can this be done by touching at the configuration files and rebooting ?

joanila, 2022-01-05:
Of course that it helps !

1- in var/log/boot.log: ---- booting motionEyeOS 20200606 ---- According to your email, this is the latest 'normal' version
2- Since I do not plan to travel to my second home till february or march, I will not update now by now the software (I will update then, even that is Spain we say that if something does work, better not touching at it)
3- I have changed in one of the cameras the image orientation (from 180 to 0) following your indications. How can I force to take a snapshot if there is nothing in the image that changes ?

Best regards

starbasessd, 2022-01-05:
Continuous snaps (like every minute) or manual snaps?

starbasessd, 2022-01-05:
1\) and 2\) :  No need to update unless you are having issues dev20201026 might fix.

joanila, 2022-01-14:
> A couple of different ways you might be able to access the MotionEyeOS Pis:
....
> > If you are running a desktop on the reverse ssh Pi, you could port forward the x session and run the WebGUI on it.

Yes, a desktop is running on my reverse ssh Pi. I have followed quite a similar approach to what you suggest: I have tunnelled VNC to the Raspi acting as gateway to my remote local network. I have accessed this Raspi through VNC, and once the VNC session open on my remote Raspi, I have opened Chromium web browser, entered the local addresses of the 2 Raspis with MotionEyeOS, opened their web interface, and been able to change the respective MotionEyeOS parameters.

As a secondary benefit, when accessing the MotionEyeOS units, I have been able to watch "live" the 2 cameras.

joanila, 2022-01-14:
Calin,

I have been able to to see live video by tunneling VNC (see the post in Github).

When accessing the 2 cameras I have seen that one of them did not show any image when accessing its webgui, while the other one did show it (see images enclosed). Just a few days ago, the images coming from the “faulty" camera showed strange color bars (see file enclosed). How can I know why the faulty camera does not show the live image ? I understand that that if it does not show the live image, it will not monitor neither any image change…

Thanks and best regards

Joan

starbasessd, 2022-01-14:
@joanila I am not Calin. I do support as partial payback for this excellent NVR package he wrote. 
I do not see any image(s) attached.
I would check to see if there are any issues with storage space, especially if they are storing locally. You can ssh to your 'supervisor' Pi, then ssh from there to the 2 camera Pis. When you do so, I would run 3 commands:
>df -h

looking at how much space shows for /, /boot, and /data on each and
>vcgencmd get_throttled

and if it reports anything but 0x0 post that number here and 
>vcgencmd get_camera 

to make sure it shows Supported = 1 and Detected = 1

If you are using the PiCams they are known to have their own issues.
In the WebGUI (after confirming the drives aren't full) you can do a reboot
too, and see if that helps.

joanila, 2022-01-14:
Following your suggestions:

***@***.*** ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    180.4M     17.6M  91% /
devtmpfs                180.3M         0    180.3M   0% /dev
tmpfs                   184.5M     16.0K    184.5M   0% /dev/shm
tmpfs                   184.5M      1.1M    183.4M   1% /tmp
/dev/mmcblk0p1           29.9M     13.3M     16.6M  44% /boot
/dev/mmcblk0p3           28.1G      1.1G     25.6G   4% /data
/dev/mmcblk0p3           28.1G      1.1G     25.6G   4% /home/ftp/sdcard
/dev/mmcblk0p3           28.1G      1.1G     25.6G   4% /home/ftp/storage
overlay                  28.1G      1.1G     25.6G   4% /usr
overlay                  28.1G      1.1G     25.6G   4% /var/log
overlay                  28.1G      1.1G     25.6G   4% /var/lib

***@***.*** ~]# vcgencmd get_throttled
throttled=0x0

***@***.*** ~]# vcgencmd get_camera
supported=1 detected=1

After rebooting, now I do get a live image, but with several abnormal lines (see file enclosed). Is there some problem with the camera ? If so, how can I check which are the problems ?

Thanks and  best regards

Joan

PS In a couple of weeks I must travel to the camera site. I will change the camera because I guess that it is faulty (even that it has worked ok for more than a year)

starbasessd, 2022-01-14:
If you are replying via email, github has a tendency to strip the attachments. If you could find at the bottom a link to the github issue (view it on GitHub) and paste it there, it would be helpful.

joanila, 2022-01-14:
I have uploaded the missing picture files

starbasessd, 2022-01-14:
I have no idea what is going on with the red and blue dots image.
When you get on site, I would consider using the dev20201026 image, at least testing it. I would not upgrade it while I was 600 km away.

joanila, 2022-01-15:
The strange thing about the red and blue dots and lines is that there are no image change alarms generated by these changing patterns....