motioneye-project/motioneyeos #2863: I have been trying to map my network share from a Synology NAS

mstewart81, 2022-01-15:
Hi 
I have been trying to map my network share from a Synology NAS the location of the folder is /volume1/surveillance
i seem to have tried everything but the correct path, 
i have attached the motioneye.log 
please can you advise

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7875796/motioneye.log)

starbasessd, 2022-01-15:
Your log indicates you have a bad path.

If up are using motionEyeOS, then:
If you are trying to save via Settings, File Storage, Storage Device, Network Share, the 
Network Server is the IP address of the SMB Server
SMB Protocol Version is whichever version your NAS supports.
Share name would be (probably) /surveillance but might be /volume1/surveillance
Username and Password as appropriate
Root Directory is usually /

If you are using Upload Media Files, then you need the FTP or SFTP settings for your NAS.
Similar as above.

If you are using motionEye on another OS, then look at how to map drives using fstab for your particular OS.

mstewart81, 2022-01-15:
tried changing to as stated /surveillance & as per image i have rooted to Guest so it doesn't need a username, although i have tried both does it come from the camera IP or the Motioneye IP for the Guest to the NAS?
i have updated the log 
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7875877/motioneye.log)
[image]

starbasessd, 2022-01-15:
motionEye IP address.

starbasessd, 2022-01-15:
It's not https:// it has to be smb://
for a Samba/SMB/CIFS Share

starbasessd, 2022-01-15:
motionEye prefers using JUST the IP address for the server.

mstewart81, 2022-01-15:
still having the issue can't map network share any other ideas?
[image]

starbasessd, 2022-01-15:
JUST the ip address. 
Username and password
motionEye doesn't understand guest with no password...

mstewart81, 2022-01-15:
have updated username & password now but still not, have given full access to the user surveillannce including applications
but still same error
[image]

mstewart81, 2022-01-15:
changed to a custom path and got it working

closed 2022-01-15 by mstewart81

mykgroup, 2022-02-02:
Hi  masterwart81 couldyou share a picture of that custom path? I'm having the same issue storing recordings in my Synology using smb. Nothing works imusing a pi zero with Motioneye OS

starbasessd, 2022-02-02:
[image]
I use motionEyeOS dev20201026 for my PiZeroWs.
You may need to adjust SMB Protocol Version depending on your Synology NAS. Start at 1.0, Click Apply, then test, Repeat by increasing Version until you get Success.
[image]
I also suggest you start at the Base of the Share, and slowly move up. As above, the Base name of the share is 'Share_Drive1', then the next folder is 'motioneye' on my NAS. Both work fine, with the above settings.

mstewart81, 2022-02-02:
i find Synology Surveillance Station better to use for my situation so no longer using Motioneye, it was a toss up for me i had 2 cameras on both but Synology clipped the post

mykgroup, 2022-02-03:
It worked. thanks. Although I might use Synology surveillance It looks better initially.