motioneye-project/motioneyeos #2864: NoIR Camera Interference after IR Blasters turn on

Xadro3, 2022-01-17:
### Preliminary Docs
I have read all the docs

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: RPI 0 with the camera ribbon, i think thats 1.3
### Camera

I am using the following type of camera: MMAL

My camera model is: Generic IR camera with IR LEDS and a OV5647 sensor in the middle

My motionEyeOS unit is connected to the network via:Wifi Dongle

I am using the following peripherals that I consider relevant to this issue:

NETGEAR WG111 GR wifi dongle
2.5A HTC charger as PSU

### Issue
My issue arises after the IR LEDs are turning on, after a few minutes i get horrible interference on my stream, i have reseated the cables multiple times, but as the camera ribbon connector on my Pi is damaged i can not rule it out 100%.
While the IR LEDs are off i have no issues whatsoever.
[image] 
[image]

### Log Files

I consider the following log files relevant to this issue:
 
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/7883109/motioneye.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/7883110/boot.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/7883111/messages.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/7883112/dmesg.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/7883115/motion.txt)

starbasessd, 2022-01-17:
Your messages log cuts off just before the time shown in the snapshot above. Can you either extend the log, or give another time that it happened to compare the log?

starbasessd, 2022-01-17:
RPi0 v1.3 is camera but no wifi. V1.1 is camera with wifi. Version is on the back.

Xadro3, 2022-01-17:
throttled=0x0

is what i get, i have the WiFi Dongle and the camera connected, the camera comes with IR LEDs so those are not seperate, operated by sensores on the board, no GPS, can i update the FW without flashing? because i mounted it real final.... (very smart yes...)

starbasessd, 2022-01-17:
It can have issues, but if you want to try:
Settings, Expert Settings, Enable Prereleases
Apply at the top
Settings, General, Software Update, Check.
Should prompt you to update to dev20201026

Xadro3, 2022-01-17:
I've been updating for the last 1.5h, should it take that long? i dont have webserver access anymore but ssh is still fine

starbasessd, 2022-01-17:
No, it should have rebooted after a few minutes. You might try command
> reboot

but I'll bet something got corrupted.

Xadro3, 2022-01-17:
reboot aint doing much :D

Xadro3, 2022-01-18:
Roller back to latest stable version and changed over to a different usb cable, now working as intended.

closed 2022-01-18 by Xadro3