motioneye-project/motioneyeos #2865: ERROR: gdrive: failed to obtain credentials: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)>

mhndr, 2022-01-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: raspberry pi zero w

### Camera

I am using the following type of camera: V4L2

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:
1970-01-03 10:07:57: [motioneye]    ERROR: gdrive: failed to obtain credentials: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)>
1970-01-03 10:07:57: [motioneye]  WARNING: accessing gdrive failed: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)>

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

mhndr, 2022-01-17:
while trying to connect to my google drive I kept getting "Accessing upload service failed" error from the webui. 
On checking the logs I found that its related to ssl certificates. Can you let me know how to fix this or if there's a workaround.
would really appreciate any help

starbasessd, 2022-01-17:
Your date in the log snippet is bad, which will prevent validating the certificate. Certificates have a not good before, and a not good after date built into them. Get your date and time set correctly and you should be good to go. Also, please post your motionEyeOS version (found in the Boot Log, or in Settings, General Settings in the WebGUI

starbasessd, 2022-01-18:
What did you find? Is the time/date being set correctly?

mhndr, 2022-01-18:
Yes I was able to set the date and time. The upload is working fine now. Thank you. 
Btw I was also wondering if I could somehow upload to two google drives at once. 
so that I could have it upload to my personal drive and another which I can share with my family.
I'm assuming I'll have to change some python scripts to get that working. could you point me to the 
right script?

closed 2022-01-18 by mhndr

starbasessd, 2022-01-18:
No, but you can share only specific folders with specific google users, that is a google function:
https://support.google.com/drive/answer/7166529
or to non google users:
https://support.google.com/drive/answer/6033939?hl=en

travdos, 2022-01-23:
Hi guys, 

has anyone figured out how to apply an SSL cert to motioneyeos. I have enabled portforwarding and am using a domain to access the motioneyeos. I need to enable SSL certificate, does anyone have any ideas on how to achieve this ?

starbasessd, 2022-01-23:
Short answer: Use [NGINX](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx)