motioneye-project/motioneyeos #2866: Question: How do I improve the video quality of my motioneye screen

jtmpush18, 2022-01-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version:
motionEye Version | 0.42.1
Motion Version | 4.1.1
OS Version | Debian 10

### Board Model

I am using the following board/model: Odroid HC4

### Camera

I am using the following type of camera: IP Camera: Amcrest IP IP3M-HX2W

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
I am attaching the following:
Screenshot from Camera: 
[image]

Screenshot from MotionEye: (approx. same area)
[image]

Using RTSP feed from camera, and the following video config in MotionEye:
[image]

Video config from Camera for main stream:
[image]

starbasessd, 2022-01-18:
What I would do (and do for my RTSP cameras on both RPis (RPi4-4GB and 8GB, RPi3B+) and Intel i5 production machine):
Lower resolution to 1280x720 at camera and local
Lower framerate to 10 FPS
You _might_ be able to go faster/higher resolution if using a 1GB Ehternet connection for everything (cameras & boards), but unlikely.
It is a matter of bandwidth and processing power, usually with that level of pixelation, the board cannot keep up with the incoming stream. 
There are some work-arounds, at the cost of some features. You can turn on Movie Passthrough (and lose text overlays, privacy masking and such), and process a lower res video for motion detection. There may be options in motion itself ( [docs](https://motion-project.github.io/motion_config.html) to help along those lines, but you are using an older version motion. You could shift to motionEye on another OS, like Debian or Ubuntu ( [instructions here on the right](https://github.com/ccrisan/motioneye/wiki/Installation) ) and get the functionality of motion 4.3.2 or 4.4.0 which may also help.

jtmpush18, 2022-01-18:
I don't seem to be able to upgrade to those versions of motion:
[image]

I am using Debian 10 - am I missing some packages?

starbasessd, 2022-01-18:
Motion versions available:
https://github.com/Motion-Project/motion/releases/
If you are using 32 bit Debian:
https://github.com/Motion-Project/motion/releases/download/release-4.4.0/buster_motion_4.4.0-1_armhf.deb
Install using:
> sudo dpkg --install \<path_to_file\>\\buster_motion_4.4.0-1_armhf.deb

If it complains about missing needed files run:
> sudo apt-get install --fix-broken

You would then have 4.4.0 of motion. I personally prefer [4.3.2](https://github.com/Motion-Project/motion/releases/download/release-4.3.2/buster_motion_4.3.2-1_armhf.deb) on my Debian or Debian derived systems, though.

starbasessd, 2022-01-18:
Storage is handled by the base OS. 
If you want to store on another device, like a NAS, then you would use the Upload Media Files
Or use /etc/fstab to mount the share to a path locally, and use Custom Path to point to the share folder. 
If you want a camera to directly store on your Odoid running Debian, you would have to install whatever file server service(s) it needs, like FTP, SFTP, SCP, NFS, CIFS/SMB, whatever.  You would then share a folder to that service. You could also use that same folder path by selecting Custom Path.

starbasessd, 2022-01-18:
Did you change resolution  and frame rate on both the camera and in motionEye?

jtmpush18, 2022-01-18:
1. Changed both Camera and Motion
Motion:. 
[image]
Camera:
[image]

2. I've got the NAS setup (with samba, cifs/smb, ftp, etc....). My backup plan is that if I can't render a good image on this Odroid HC4, then I will move motion to run on a "better" machine, and I'll use the network storage to put the motion files on it.
3. is that package/deb file for armbian 64bit also?
Thanks for the responses.

starbasessd, 2022-01-18:
It may be the camera is pushing a bitrate that the Odroid can't support. 
Try lowering the quality and/or bitrate down by half at stages (1024/512/256 and see if either affect it.
I don't know 'armbian', but there are multiple ARM versions available:
4.4.0 [ARM64](https://github.com/Motion-Project/motion/releases/download/release-4.4.0/buster_motion_4.4.0-1_arm64.deb), [ARMEL](https://github.com/Motion-Project/motion/releases/download/release-4.4.0/bullseye_motion_4.4.0-1_armel.deb) and [ARMHF 32bit](https://github.com/Motion-Project/motion/releases/download/release-4.4.0/bullseye_motion_4.4.0-1_armhf.deb)
4.3.2 [ARM64](https://github.com/Motion-Project/motion/releases/download/release-4.3.2/buster_motion_4.3.2-1_arm64.deb), [ARMHF 32bit](https://github.com/Motion-Project/motion/releases/download/release-4.3.2/buster_motion_4.3.2-1_armhf.deb)

cap9qd, 2022-01-18:
I had a lot of issues with my cameras and stability trying to get my HC1/HC2 working reliably under Ubuntu 20.0.4/Motioneye/motion. With Motioneyeos I have not had many issues and run 3 cameras at 2560x1920 reolink, one iqeye at 720p, 2 Dahua at 1080p and one esp32camera at something small 480p. I run them all as passthrough camera except the esp32cam. All are running at 6fps. I am bumping the ram limitation of the baord since it's only 2gb RAM. Is the HC4 derived from the XU4? The HC1/HC2 boards were (no EOL by Odroid unfortunately) the same processor and image as the XU4. 
[image]

jtmpush18, 2022-01-18:
Thanks for commenting.
The Odroid XU4 and the HC4 are compared in the following chart:
[image]
[image]
[image]

I'm working on modifying the config in the camera and in motioneye to get an unpixellated picture.

jtmpush18, 2022-01-18:
Well, I finally fixed my issue - the way that they taught me (years ago) in Engineering school.
For anyone who reads this at a later date, and has this same issue:
1. remove the entire camera feed (use the garbage can icon).
2. put the same camera details back.
3. Voila - no pixellation.

Thanks to everyone who help me, especially 
starbasessd

closed 2022-01-18 by jtmpush18

starbasessd, 2022-01-18:
> 
> Question: (Please forgive my Linux ignorance, I'm a newbie at this). Do you know what I'm doing wrong here: [image]
you need to get the install (deb) file local, 
> wget https://github.com/Motion-Project/motion/releases/download/release-4.4.0/buster_motion_4.4.0-1_arm64.deb 

is what I usually use. 
Glad to hear the 'simple' answers work. Yes, I forget, too.