motioneye-project/motioneyeos #2869: Jawsper has a New build of motionEyeOS available for PiZero2W

starbasessd, 2022-01-19:
Can you try this new version? https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev
He updated rpi-firmware (not sure if needed) and rpi-wifi-firmware (this may fix your issue).

Originally posted by @jawsper in [2265](https://github.com/ccrisan/motioneye/issues/2265#issuecomment-1016649578)
There appears to be 2 versions of the PiZero2W in the wild, now with different WiFi chips.

orcadiver, 2022-01-23:
Hi all, i just installed the mentioned version and it works except for the highest resolution of 1920x1080?!
(And updating motionEyeOS directly over the running system will cause a system crash? ;-).
...Raspberry Pi Zero 2 W | Raspberry Pi NoIR Camera V2.1 | Sandisk Extrem PRO 128GB

starbasessd, 2022-01-23:
@orcadiver Yes, it would crash, since it's not linked into the 'official' images, and couldn't be upgraded. I don't know if @jawsper linked his older image to his newer image in the repository or not. 
What framerate are you trying to use? If over 30FPS, try slowing to 15FPS or less.

orcadiver, 2022-01-23:
@starbasessd
I can generally only set frame rates up to 30. For Full HD, there is an error with each one.
Is it possible to match the colours somewhere (white point)?
(Currently it has a violet/red cast with LED artificial light [4000k]).
Is there likely to be an official version of motionEyeOS for the Raspberry Pi Zero 2 W?

starbasessd, 2022-01-23:
Sounds like the NOIR issue [here](https://pimylifeup.com/raspberry-pi-camera-vs-noir-camera/)
I would slow it to 5FPS and if that works, slowly move it up.
Next official build should be good.

orcadiver, 2022-01-23:
Is it known if an official firmware is being worked on?
Thanks for the tip "NOIR" - I have now ordered two without it.

starbasessd, 2022-01-23:
What do you mean 'official firmware'?
motionEyeOS uses what's available at the time of the build from RPiFoundation...

orcadiver, 2022-01-23:
from RPiFoundation

orcadiver, 2022-01-23:
For them it shouldn't be so complicated to adapt an image?

starbasessd, 2022-01-23:
BTW, due to the Bullseye fiasco with releases that will no longer support the familiar raspicam apps and Picamera Python library, it may be built on Buster instead of moving forward to Bullseye, especially if what they (RPiFoundation) applied as a 'fix' is the best they want to do... 
motionEye is built on ThingOS and buildroot, not the full OS. It's more a matter of buildroot and ThingOS adapting what RPiFoundation makes available. Oh, and BTW, lately RPiF hasn't been very forthcoming in making changes (especially to PiZero2W) on the fly, updating RPiOS and forgetting to mention the changes. (e.g. the Wifi chip change on the board breaking stuff like Jawsper's first image...

orcadiver, 2022-01-23:
If such projects are not only to have a tinkering status, a halfway stable roadmap would be useful - at least I would expect a statement whether and which new Pi's will get support. And the Zero 2 W is not really new anymore.

starbasessd, 2022-01-23:
Think about the hundreds of sites (thousands?) that wrote 'educational' software based on the old Buster Picamera and raspicam stuff, and suddenly RPiF dropped all support for it (including the lessons THEY wrote) without any serious updates. When the complaints started rolling in, they 'bactracked' a little, but it's too little too late. Now everyone is suggesting not to update to Bullseye while scrambling to get fixes in place. sigh...

orcadiver, 2022-01-23:
Jo, really bitter. I often can't understand what people are thinking strategically. Then you don't have to be surprised when the community leaves in a huff. A clear concept is an indispensable basis not only in the paid business - but what am I saying, even there it is often lacking.

orcadiver, 2022-01-23:
Let's hope that there will be a completely functioning image in the near future - otherwise my Bird-Cam project will soon be dead again - unfortunately.

orcadiver, 2022-01-23:
After years of being annoyed with shortcomings of Axis cameras, I thought a Raspi variant would be a good alternative...

starbasessd, 2022-01-23:
You can switch to the motionEye on another OS using Buster [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) to get the 'latest' OS updates. If you weren't using a CSI PiCam, you could go with the Bullseye install. Either install would get you a newer motion version, updated ffmpeg, etc... You'd lose Fast Network Camera, have to set up your own networking and shares, etc, though...

orcadiver, 2022-01-23:
Sounds like good alternatives - thanks for these tips!

DL7AG-Berlin, 2022-01-26:
I have the problem that the version always causes a reboot, that after a few seconds....er does not find a wifi adapter and some more.

starbasessd, 2022-01-26:
Jawsper has 2 images available (we know of 2 versions of the PiZero2W) did you try the [dev one here?](https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev)

DL7AG-Berlin, 2022-01-26:
Under your link is only one? I have also tried these.

starbasessd, 2022-01-26:
That is an 'unofficial' build that people are using. Can you give all the info on the back of your PiZero2W, please?
https://user-images.githubusercontent.com/592762/150166675-28778432-11f2-418e-b7ed-96322813dadc.jpg
Just to the right of the raspberry in the top photos, there's a pair of numbers. It would be helpful to know what you have. In the example on the left, the numbers are 2121 &  1.5. on the right, they are 2821 & 2.11.

DL7AG-Berlin, 2022-01-26:
I have now tried it on three different Zero2W. 
It says "wired network not device" and then reboots.
I use the version....this [https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev](url)
Am I doing something wrong?

jawsper, 2022-01-26:
Make sure you follow these instructions for setting up wpa_supplicant: https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration (for the Windows instructions go straight to step 5)

DL7AG-Berlin, 2022-01-26:
> That is an 'unofficial' build that people are using. Can you give all the info on the back of your PiZero2W, please? [...]

This is mine... https://s20.directupload.net/images/220126/5ct5vu6g.jpg

starbasessd, 2022-01-26:
Also, if using Raspberry Pi Imager, do NOT use their presets for any settings (under CTRL SHIFT X)

starbasessd, 2022-01-26:
@jawsper looks like a 3rd version of the PiZero2W.

jawsper, 2022-01-26:
@DL7AG-Berlin have you tried this older build? https://github.com/jawsper/motioneyeos/releases/tag/20211113

orcadiver, 2022-01-26:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=DE
   
network={
    ssid="abcdef"
    psk="abcdefghijklmnop"
    key_mgmt=WPA-PSK
}

orcadiver, 2022-01-26:
Disable WiFi configuration via imager and enable skip setup wizard!
After writing the image to the SDCard on the PC (Windows) eject it and plug it in again, copy the said CONF file into the root directory of the card and then plug it into the Pi and boot.

orcadiver, 2022-01-26:
works with 2821 | 2.13

DL7AG-Berlin, 2022-01-26:
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev update_config=1 country=DE
> 
> network={ ssid="abcdef" psk="abcdefghijklmnop" key_mgmt=WPA-PSK }

THAT had been it!
With the new Opa_supplicant.conf it works!
MANY THANKS - the examples that otherwise exist on the net did not work.

starbasessd, 2022-01-26:
You do NOT want to "enable skip setup wizard", either. You do not want to have it make any changes, as motionEyeOS uses a 3 partition system, with /boot and / (root) (Partitions 1 and 2 being read-only) and /data being read-write with file system overlays.

orcadiver, 2022-01-26:
Be careful with high camera resolutions (1.920x1.080) - these cause the system to freeze (probably not for USB cameras) and can only be fixed with a new image (in my case).

Nypan-sr, 2022-01-27:
> @jawsper looks like a 3rd version of the PiZero2W.

And i have two with 0621  3.20 & 0621  4.15

jawsper, 2022-01-27:
Starting to get the feeling these may not all be distinctly different revisions anymore... Doesn't seem to be a date code either ...?

jawsper, 2022-01-27:
Ah perhaps the first number is WWYY, week number and year of production. That seems to match the different values that people have posted here so far.

orcadiver, 2022-01-27:
Wait, there was already an error before that...

root@motionEye-Cam2:~# apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ffmpeg is already the newest version (7:4.3.2-0+rpt3+deb11u2).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  mariadb-common mysql-common
The following NEW packages will be installed:
  libmariadb3 libmicrohttpd12 libpq5 mariadb-common mysql-common
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 462 kB of archives.
After this operation, 1,473 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf mysql-common all 5.8+1.0.7 [7,464 B]
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf mariadb-common all 1:10.5.12-0+deb11u1 [36.3 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libmariadb3 armhf 1:10.5.12-0+deb11u1 [160 kB]
Err:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libpq5 armhf 13.4-0+deb11u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Get:4 http://mirror.de.leaseweb.net/raspbian/raspbian bullseye/main armhf libmicrohttpd12 armhf 0.9.72-2 [98.0 kB]
Fetched 301 kB in 2s (135 kB/s)
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-13/libpq5_13.4-0%2bdeb11u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@motionEye-Cam2:~# wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb
--2022-01-27 15:45:44--  https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/20822620/37d72a00-1781-11eb-801c-ff7902d43cc5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220127%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220127T144419Z&X-Amz-Expires=300&X-Amz-Signature=ebe06b41aeb8fa743f0814267c4caef89ac60215d949f095e0b351486e4c87ec&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=20822620&response-content-disposition=attachment%3B%20filename%3Dpi_buster_motion_4.3.2-1_armhf.deb&response-content-type=application%2Foctet-stream [following]
--2022-01-27 15:45:44--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/20822620/37d72a00-1781-11eb-801c-ff7902d43cc5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220127%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220127T144419Z&X-Amz-Expires=300&X-Amz-Signature=ebe06b41aeb8fa743f0814267c4caef89ac60215d949f095e0b351486e4c87ec&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=20822620&response-content-disposition=attachment%3B%20filename%3Dpi_buster_motion_4.3.2-1_armhf.deb&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 485892 (475K) [application/octet-stream]
Saving to: ‘pi_buster_motion_4.3.2-1_armhf.deb’

pi_buster_motion_4.3.2-1_armhf.deb      100%[============================================================================>] 474.50K  --.-KB/s    in 0.1s

2022-01-27 15:45:44 (3.76 MB/s) - ‘pi_buster_motion_4.3.2-1_armhf.deb’ saved [485892/485892]

starbasessd, 2022-01-27:
@orcadiver : Will check current installability on RPiOS Bullseye, momentarily.
@jawsper  ; We know, it's all over the place, but this last issue IS related to motioneye, and I'm checking. He highjacked this issue, and after I get it resolved, I will close and lock, and restart it with a pin.

starbasessd, 2022-01-27:
Clean install of RPiOS Bullseye (10/30/2021) (Bullseye) (Desktop) Fully updated on Pi3B+.
Following install instructions here:
[Install on RPiOS Bullseye](https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye)
No Issues found, started normally, mounted Network Camera normally. Updated comments in instructions to indicate new last tested date.
Am closing this issue, and locking. If you have issues after starting over, please open a new issue.

closed 2022-01-27 by starbasessd