motioneye-project/motioneyeos #2870: Protocol "rtsp" not supported or disabled in libcurl

Brutus11111, 2022-01-20:
Arch Linux install
```
Linux motioneye 5.16.1-arch1-1 #1 SMP PREEMPT Sun, 16 Jan 2022 11:39:23 +0000 x86_64 GNU/Linux
```

Running in a VM. 

### motionEyeOS Version

I am running motionEyeOS version: current one from github

### Board Model

Intel Xeon

### Camera

Network Camera
Wyze v1 

### Network Connection

Ethernet

Ok so here's the problem. There appears (I think) to be something going on with curl not accepting rtsp. It appears (according to my google search) to be related to this:

https://github.com/curl/curl/pull/4750

It gives me an error that libcurl doesn't support RTSP (see title of issue). So I tried to downgrade RTSP, to version 7.63. Now motion eye won't start because I downgraded the package to a version that should support RTSP.

```
Jan 20 23:11:40 motioneye dbus-daemon[314]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.home1.service': Unit dbus-org.freedesktop.home1.service not found.
Jan 20 23:11:40 motioneye sudo[346]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Jan 20 23:11:40 motioneye kernel: audit: type=1101 audit(1642716700.376:58): pid=346 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="rijd" exe="/u>
Jan 20 23:11:40 motioneye sudo[346]:     rijd : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/bin/bash
Jan 20 23:11:40 motioneye audit[346]: CRED_REFR pid=346 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hos>
Jan 20 23:11:40 motioneye sudo[346]: pam_unix(sudo:session): session opened for user root(uid=0) by rijd(uid=1000)
Jan 20 23:11:40 motioneye audit[346]: USER_START pid=346 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/>
Jan 20 23:11:40 motioneye kernel: audit: type=1110 audit(1642716700.379:59): pid=346 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct=">
Jan 20 23:11:55 motioneye systemd[1]: Started motionEye Server.
Jan 20 23:11:55 motioneye audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=motioneye comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? re>
Jan 20 23:11:55 motioneye kernel: kauditd_printk_skb: 1 callbacks suppressed
Jan 20 23:11:55 motioneye kernel: audit: type=1130 audit(1642716715.578:61): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=motioneye comm="systemd" exe="/usr/lib/systemd/systemd" host>
Jan 20 23:11:55 motioneye meyectl[354]: Traceback (most recent call last):
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/bin/meyectl", line 8, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     sys.exit(main())
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 254, in main
Jan 20 23:11:55 motioneye meyectl[354]:     import server
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 31, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import handlers
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 30, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import config
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 32, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import motionctl
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 28, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import mediafiles
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 41, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import uploadservices
Jan 20 23:11:55 motioneye meyectl[354]:   File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
Jan 20 23:11:55 motioneye meyectl[354]:     import pycurl
**Jan 20 23:11:55 motioneye meyectl[354]: ImportError: pycurl: libcurl link-time version (7.63.0) is older than compile-time version (7.81.0)**
Jan 20 23:11:55 motioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jan 20 23:11:55 motioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.
Jan 20 23:11:55 motioneye audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=motioneye comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res
```

Maybe I'm missing something.

starbasessd, 2022-01-20:
Curl doesn't support RTSP://, it is used to copy files from the internet to the local machine. 
Which install instructions are you using?
The one at [Install on Arch](https://github.com/ccrisan/motioneye/wiki/Install-On-Arch) I tested on the 27th of Dec 2021, and it worked fine.

starbasessd, 2022-01-20:
BTW, this is motionEyeOS, you are looking for support for [motionEye](https://github.com/ccrisan/motioneye/)

Brutus11111, 2022-01-20:
> Curl doesn't support RTSP://, it is used to copy files from the internet to the local machine. Which install instructions are you using? The one at [Install on Arch](https://github.com/ccrisan/motioneye/wiki/Install-On-Arch) I tested on the 27th of Dec 2021, and it worked fine.

That's the installation method I tried. I'll reinstall the VM and try it again tomorrow.

Brutus11111, 2022-01-20:
> BTW, this is motionEyeOS, you are looking for support for [motionEye](https://github.com/ccrisan/motioneye/)

Thanks, if reinstallation doesn't work, I'll ask them. Thank you!

closed 2022-01-20 by Brutus11111