motioneye-project/motioneyeos #2871: An error occurred. Refreshing is recommended.

gijsbreght, 2022-01-24:
Hi,
I've been using MotionEyeOS for a few years.
Since this week I notices I can't add camera's.
And when I want to change something in the settings I get a error as respons.
"An error occurred. Refreshing is recommended.

In the log file I can see :
IOError: [Errno 28] No space left on device

The external disk for movies has enough space.
MotionEye OS itself...idk.... I would think there is enough space on the SD.
How can I solve this issue?

Friendly greeting,

Gijsbreght

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS V

I am running motionEyeOS version: 0.42.1.

### Board Model

I am using the following board/model:  Raspberry PI 4.

### Camera

I am using the following type of camera: 4 Network Camera and  1 Simple MJPEG Camera).

My camera model is: Different Axis and Sonoff..

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[motioneye (5).log](https://github.com/ccrisan/motioneyeos/files/7928324/motioneye.5.log)

starbasessd, 2022-01-24:
I need the version of motionEyeOS, it will be in a date format. You posted the version of motionEye (0.41)
Can you log in to the console, or via SSH and report the results of:
df -h
In particular need to find the space available on / (root) and /data

gijsbreght, 2022-01-24:
[image]
[image]

gijsbreght, 2022-01-24:
Version motionEyeOS 20200606

starbasessd, 2022-01-24:
/data is full. that is where all the config files are stored, in /data/etc in the motioneye.con, motion.conf and camera-x.conf files. I don't know what you need to keep, but I would try to move any *.jpg and *.mp4 files over to the /home/ftp/storage folder...
Do not use cp which is copy, use mv which is move.

gijsbreght, 2022-01-26:
Thanks! You helped me out!

closed 2022-01-26 by gijsbreght