motioneye-project/motioneyeos #2872: MotionEyeOS is only recording one video in continuous mode and none after that.

scott8, 2022-01-24:
Steps in setup of MotionEyeOS system

sed Raspberry Pi Imager v 1.6.2 on Ubuntu 20.04.3 LTS
Wrote motioneyeos-raspberrypi4-20200606.img.xz and Verified it to Micro SD Card 16GB, SanDisk Ultra.
Camera: https://www.amazon.com/gp/product/B07RGKN6QT
booted RPi 4B
Web browser Logged in as admin
Camera is discovered automatically as MMAL Camera 640x480 resolution.
Network connection is Ethernet.
I have video and keyboard attached.

Changed Settings:
Motion Detection: Off
Movies:
  Recording Mode: Continuous Recording
  Maxmimum Movie Length: 60
  Preserve Movies: For One Week
Preferences:
  Layout Columns: 1
General Settings:
  Admin Password and Surveillance Password set.
  Time Zone: US/Pacific
  Hostname: MotionEyeOS1
Services:
  FTP: Off
  Samba: Off
Expert Settings:
  NTP Server: clock.isc.org
APPLIED Changed Settings.
RPi Rebooted.
Logged into MotionEyeOS using new password.
Date & Time are now set correctly.

Clicked on recorded movies icon.
There is one. It's 12-35-09.mp4.
It has no preview.
It's 1:09 long and plays correctly in the browser.
There are no other movies recorded after 5 minutes.

Rebooted RPi with full shutdown. 
One more recorded movie showed up with 1:09 length.
Downloaded Logs.
Submitted this ticket.
[boot.log](https://github.com/ccrisan/motioneyeos/files/7928817/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/7928818/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/7928819/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/7928820/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7928821/motioneye.log)

starbasessd, 2022-01-24:
Known issue with 20200606. Use dev20201026 (I personally know it works & recommend) or downgrade to 20190911 (my last known good working for that function).

scott8, 2022-01-24:
I installed the motioneyeos-raspberrypi4-dev20201026.img.xz, booted it.
Logged in as admin.
The entire left side configuration panel is greyed out. I can't change
anything.

Scott

scott8, 2022-01-24:
I deleted the camera and added it. Now the left panel is working.

scott8, 2022-01-24:
I made the same changes and Applied them.
After the reboot, some or all the changes did not get applied.
The left panel is again greyed out, after I log in as admin with the new
password!
This dev version is apparently broken.

starbasessd, 2022-01-24:
It may be a bad copy, dev20201026 is my default for all Pis (each with it's own image of course).
Which browser are you using?
Have you cleared cache & cookies?
Do not Select option "Remember Me" until all settings are remedied. 
Can you post a screen shot, and state what you changed (not necessarily what you changed it to, if security related). 
Do NOT change user or admin passwords until last. 
Change 1 thing at a time, apply, log out, log back in, and make the next change.
Report where it breaks for you.

scott8, 2022-01-24:
I'm using Chrome on Linux.
It works great with the production version of MotionEyeOS.
I reinstalled the dev version to change one thing at a time:
Booted it up.
Logged in as admin.
The camera was auto-detected and working.
The left panel is all greyed out so I can't make any changes.
I deleted the camera and added it again.
The left panel is functional as normal.
I turned off Motion Detection. It's the only thing I changed.
The APPLY button does not show up.

Scott

scott8, 2022-01-24:
I just cleared cache and cookies, etc. Now the page is working nicely.

scott8, 2022-01-24:
Now I changed the settings again to what I want and continuous recording
has no videos saved at all. I set the max length to 60 seconds.

Scott

scott8, 2022-01-24:
I rebooted it.
Now I get video every 60 seconds but the more recent ones are coming back
with this:
"Error: Media format unsupported or otherwise unavilable/unsuitable for
playing."
... and are not viewable.

Scott

Angryestbeard, 2022-01-24:
Might be a bit late but I have been having the same issue with motionos on a pi4 4g.
After much reading I tryed enabling pre releases under expert settings.
I am now able to record more then one file.

scott8, 2022-01-25:
It seems to be working now.
I don't know what caused it to misbehave earlier.
I'm getting recordings every 60 seconds that are usable.

scott8, 2022-01-25:
Now it's not working again. No  idea why. :(

scott8, 2022-01-25:
Now it's recording video but it's unusable.
I reduced my video resolution and this happened.

Angryestbeard, 2022-01-25:
What codec are you using under movies? Depending on the version you won't get in browser playback.
The update only took 2-3 mins tho and has sorted my issues out.

scott8, 2022-01-25:
I am using the out of the box codec. (H.264/OMX .mp4).
I switched to 10 minute movie length.
One of the movies it recorded is usable. The other isn't.
I don't know why it's being intermittent about this.

scott8, 2022-01-25:
Very strange. There must be some post-processing on the movies?
Movies that were not usable in the browser now are usable.