motioneye-project/motioneyeos #2874: Recommended images, settings, etc.

starbasessd, 2022-01-26:
RPiZero2W: 
** There are 2 'unofficial' images:
1. Latest: https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev
2. Previous: https://github.com/jawsper/motioneyeos/releases/tag/20211113

Rpi4, All Versions:
** It is recommended (and required for RP4-4GB & 8GB) to use the [dev20201026](https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz) image.

** If you want to use Continuous Recording, please use either the dev20201026 (recommended) or 20190911 images.
 
** Please check the information available in both the [motionEyeOS Wiki](https://github.com/ccrisan/motioneyeos/wiki) and the [motionEye Wiki](https://github.com/ccrisan/motioneye/wiki), as most can apply to both. 

** Please Do Not use RaspberryPiImager to preset ANYTHING in motionEyeOS. It will corrupt settings files.

Nik71git, 2022-04-02:
uh... I realize now I probaly preset some minor issues with piimager... I'll flash again following your tips. I hope motioneyeOS will not end and a new developer team can support It.

trujillo66, 2023-04-03:
Hi everyone,
Is thera an image for Radxa Rock 3A board?
Or a fork in progress to join?

starbasessd, 2023-04-03:
No.
You can do a 'build your own image' [following the instructions here](https://github.com/motioneye-project/motioneyeos/wiki/Building-From-Source)
You can install your own OS (lite version preferably) and [install the appropriate version from here](https://github.com/motioneye-project/motioneye/wiki/Installation)
Be aware, motionEye default (and motionEyeOS) are written in Python2.7 and you need to be able to install Python2.7
There is a beta [Python3.x instruction set here](https://github.com/motioneye-project/motioneye/issues/2373)

ajuyalbozza, 2025-02-24:
Help. So I got a raspberry Pi Zero 2 W, and I am trying to download motion eye OS on it. I have had a couple problems. One, whenever I write the SD card (FYI the device was set to Raspberry Pi 2 W, and the OS was the one done by Jawsper ), it disappears from my files until i remove the SD card and reinsert it, which is something you are not supposed to when preconfiguring the wifi. I still got it to work, and it plugged into my monitor. I then was able to see the IP address and even open the open eye os web page. On there, It says unable to open video device. 

If anyone could help, it would be greatly appreciated. I am new to raspberry Pi's and this kind of stuff, and don't know if the problem is software or hardware. 

Some screenshots will be attached below.

[image]
[image]
[image]

Extra info: 
Raspberry Pi 2 W. 
Using latest version from Jawsper. 
Using windows 11.
Country: Canada
Camera: Camera Module 3 (not wide)