motioneye-project/motioneyeos #2876: I can't connect to wifi using Raspberry Pi Zero W2

MrAppAndCrap, 2022-01-27:
Hey there,

So i'm using the the latest modified version of MotioneyeOS for RPI Zero W2 modified by @jawsper. 
I am trying to connect it to set up my wifi preconfiguration files but it will not connect. 

I have tried:

Connecting to my home wifi router (talk talk one).
Connecting to my Phone Hotspot.

With no luck. 

My old RPI zero w orginal connected fine to my home wifi and phone without a problem. 
(The sdcarder port broke so I can't use it anymore) 

Any tips? 

Can you see anything wrong with my config file? (I'm uk based)

       country=GB
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="AndroidAP"
            psk="Mypassword"
        }

I save it using notepad++ and encode it. It still gets seen as a text file - is that ok? 

Any help?

starbasessd, 2022-01-27:
Missing part of a line and a whole line:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
&nbsp; &nbsp; &nbsp; ssid="AndroidAP"
&nbsp; &nbsp; &nbsp; psk="Mypassword"
&nbsp; &nbsp; &nbsp; key_mgmt=WPA-PSK
}

MrAppAndCrap, 2022-01-27:
I'm using the config template suggested in the EyemotionOS wiki [](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration) So I'd hope nothings missing.

Just tried your version - still no luck

starbasessd, 2022-01-27:
Do you have a monitor & keyboard attached, or another Pi you can put the SDCard in?

starbasessd, 2022-01-27:
Unfortunately, that link points into the ThingOS Wiki, is somewhat out of date, and I don't have access to change it there. RPi Foundation [recommendations](https://www.raspberrypi.com/documentation/computers/configuration.html#adding-the-network-details-to-your-raspberry-pi) is in the format I posted here.

starbasessd, 2022-01-27:
Make sure there isn't a .txt extension (wpa_supplicant.conf.txt) in the filename, too.

MrAppAndCrap, 2022-01-27:
> Make sure there isn't a .txt extension (wpa_supplicant.conf.txt) in the filename, too.

Nope there isn't .txt in the name but is still showing as a text file.

Nope, my mini hdmi cable has gone missing. Just went looking round the house for it. I have a feeling a family member got to it so it's basicially as good as gone now lol. 

The only way I'll be able to access it it via wifi.

starbasessd, 2022-01-27:
It IS saving as a .conf.txt file. Look at the tab where you are editing it.
In Windows File Explorer Go To View at the top, and towards the right , check the box for File Name Extensions. Then take the .txt off the filename.

starbasessd, 2022-01-27:
Like I said, I wish I could. If @ccrisan gives me access to ThingOS I will, otherwise I may break that link and write a whole 'nuther page for it. If you are working, please close it.

closed 2022-01-27 by MrAppAndCrap