motioneye-project/motioneyeos #2877: Another endless reboot problem & my workaround which I assume is not correct - RPi0W

jjfelten, 2022-01-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 0W

### Camera

I am using the following type of camera: Raspberry Pi Camera Module

My camera model is:  OV5647

### Network Connection

My motionEyeOS unit is connected to the network via: Wireless Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:  None

### Log Files

I consider the following log files relevant to this issue:

/var/log/messages:  
Jan  1 00:00:05 meye-4ef5f9c2 user.notice throttlewatch: currently: ok, temperature: 41 C
Jan  1 00:00:10 meye-4ef5f9c2 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 00:00:10 meye-4ef5f9c2 daemon.info dhclient: DHCPACK from 192.168.2.1
Jan  1 00:00:11 meye-4ef5f9c2 daemon.info dhclient: bound to 192.168.2.6 -- renewal in 726818 seconds.
Jan  1 00:00:11 meye-4ef5f9c2 cron.info crond[967]: crond (busybox 1.29.3) started, log level 8
Jan  1 00:00:11 meye-4ef5f9c2 cron.info crond[970]: crond (busybox 1.29.3) started, log level 8
Jan  1 00:00:12 meye-4ef5f9c2 auth.info sshd[1001]: Server listening on 0.0.0.0 port 22.
Jan  1 00:00:13 meye-4ef5f9c2 daemon.notice proftpd[1018]: localhost - ProFTPD 1.3.6 (stable) (built Sat Jun 6 2020 13:15:40 UTC) standalone mode STARTUP
Jan  1 00:00:16 meye-4ef5f9c2 auth.info sshd[1065]: Accepted none for admin from 192.168.2.35 port 51929 ssh2
Jan  1 00:01:05 meye-4ef5f9c2 user.notice throttlewatch: currently: ok, temperature: 40 C
Jan  1 00:02:05 meye-4ef5f9c2 user.notice throttlewatch: currently: ok, temperature: 40 C
Jan  1 00:02:29 meye-4ef5f9c2 daemon.info : starting pid 1702, tty '': '/etc/init.d/rcK'
Jan  1 00:02:33 meye-4ef5f9c2 auth.info sshd[1001]: Received signal 15; terminating.
Jan  1 00:02:33 meye-4ef5f9c2 auth.info sshd[1065]: Exiting on signal 15
Jan  1 00:02:33 meye-4ef5f9c2 daemon.notice proftpd[1018]: localhost - ProFTPD killed (signal 15)
Jan  1 00:02:33 meye-4ef5f9c2 daemon.notice proftpd[1018]: localhost - ProFTPD 1.3.6 standalone mode SHUTDOWN

/var/log/boot.log:  
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: failed
---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping network: done
 * Stopping wpa_supplicant: done
 * Stopping rngd: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done

/data/etc/motioneye.log:  
1970-01-01 00:00:18: [motioneye]     INFO: hello! this is motionEye server 0.42.1
1970-01-01 00:00:20: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
1970-01-01 00:00:20: [motioneye]     INFO: cleanup started
1970-01-01 00:00:20: [motioneye]     INFO: wsswitch started
1970-01-01 00:00:20: [motioneye]     INFO: tasks started
1970-01-01 00:00:20: [motioneye]     INFO: mjpg client garbage collector started
1970-01-01 00:00:20: [motioneye]     INFO: server started
1970-01-01 00:02:32: [motioneye]     INFO: interrupt signal received, shutting down...
1970-01-01 00:02:32: [motioneye]     INFO: server stopped
1970-01-01 00:02:32: [motioneye]     INFO: tasks stopped
1970-01-01 00:02:32: [motioneye]     INFO: bye!

### The Problem:  

I followed the usual instructions to create the wpa_supplicant.conf file which worked fine and the R-Pi connected to my wireless Ethernet AP on boot.  I had also created an empty file named "ssh" to have it automatically start the ssh daemon, so I ssh'ed in only to have it reboot on me.  It reboots every 2.5 minutes.  Reading here I see people have problems with it rebooting when it cannot connect to the network, but mine connects to the network fine.  While it is running I can access the Internet from it etc.  Note the Raspberry Pi 0W has no wired Eithernet port so I am using its wireless Ethernet to ssh in.  

I read here:  https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-continuously-rebooting
" , , , You can disable the various software watching mechanism in the Expert Settings section."  
But can't stay logged in long enough to find that configuration file wherever it is.  

On this page:  
https://github.com/ccrisan/motioneyeos/wiki/Configuration
I saw a link on the right labeled "watch.conf" but it is a broken link to:  
https://github.com/ccrisan/thingos/wiki/watch.conf 

I tried commenting out the /sbin/reboot command in /etc/init.d/panic, but that didn't help.  

I tried setting "enable_reboot false" in /etc/motioneye.conf, but that didn't help.

I tried:  service watchdog stop, but that didn't help

I tried:  Editing /data/etc/watch.conf and changing:
-LINK_WATCH="true"
+LINK_WATCH="false"
-IP_WATCH="true"
+IP_WATCH="false"
-#MEYEWATCH_DISABLE="true"
+MEYEWATCH_DISABLE="true"
And after the next reboot cycle THAT FINALLY HELPED.  Now I could stay ssh'ed in at least.  However, while motioneye seemed to be running, netstat showed nothing was listening on port 80.  

I noticed /data/etc/motioneye.conf was empty so I copied:  
cp -p /etc/motioneye.conf /data/etc/motioneye.conf
And ran:  
/etc/init.d/S85motioneye start
And finally it was working.  

The reason I'm still posting this as a problem is because I doubt all this hacking was the right way to do this.  Can someone please tell me the correct way?  Why didn't it work out of the box as expected?  

Thank you.  Looks like a very nice system once it is running and not constantly rebooting.

starbasessd, 2022-01-30:
A couple of quick questions:
Have you tried dev20201026?
What size SD Card were you using, and how much free space available on /dev/mmcblk0p3 (/data)?
What is the result of the command:
vcgencmd get_throttled
???
It sounds to me like something was corrupted in the image. What was used in burning the image to the SDCard? If RPiImager, was anything set in the 'hidden' menu (ctrl-shift-x)?

starbasessd, 2022-01-30:
Also, your Pi didn't pick up date & time from the network. You might want to take a look at that, too. I do not know what you are using to do DHCP services, if you are issuing a static IP address, etc, but it is possible something is not quite right, there, too.

starbasessd, 2022-01-30:
To answer 2 other statements:
>I read here: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-continuously-rebooting
>" , , , You can disable the various software watching mechanism in the Expert Settings section."
>But can't stay logged in long enough to find that configuration file wherever it is.

Located in the WebGUI once it is up and running.
>On this page:
>https://github.com/ccrisan/motioneyeos/wiki/Configuration
>I saw a link on the right labeled "watch.conf" but it is a broken link to:
>https://github.com/ccrisan/thingos/wiki/watch.conf

Corrected: Configuration to watchdog.conf, and link to correct page.

jjfelten, 2022-01-30:
Thank you for your quick replies.  It looks like something went wrong with the initial boot as I created a new copy following the exact same steps the same way, and the second copy worked correctly.  Sorry I didn't try creating a boot microSD before posting here, but since I saw no errors the first time, and it was booting, I didn't realize it hadn't auto-configured correctly.  

I'll answer your questions about the first attempt in case that helps understand why the first attempt didn't work correctly.  Perhaps it will help someone else in the future.  

>  Have you tried dev20201026?
No.  Would have been a good idea.  

> What size SD Card were you using, and how much free space available on /dev/mmcblk0p3 (/data)?

32GB.  26.6GB free on /data.

> What is the result of the command:  vcgencmd get_throttled
[root@meye-4ef5f9c2 init.d]# vcgencmd get_throttled
throttled=0x0

> What was used in burning the image to the SDCard?:  

root@beethoven:~# wget https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz
root@beethoven:~# md5sum motioneyeos-raspberrypi-20200606.img.xz 
1764e66fc46c77a5b5279f8f0978c671  motioneyeos-raspberrypi-20200606.img.xz
root@beethoven:~# xzcat motioneyeos-raspberrypi-20200606.img.xz | dd bs=4M of=/dev/mmcblk0
0+42126 records in
0+42126 records out
346030080 bytes (346 MB, 330 MiB) copied, 19.4715 s, 17.8 MB/s

root@beethoven:~# mount /dev/mmcblk0p1 /mnt
root@beethoven:~# ls /mnt
bcm2708-rpi-b.dtb       bootcode.bin  fwupdater.gz  start.elf
bcm2708-rpi-b-plus.dtb  cmdline.txt   initrd.gz
bcm2708-rpi-cm.dtb      config.txt    kernel.img
bcm2708-rpi-zero-w.dtb  fixup.dat     overlays
root@beethoven:~# vi /mnt/wpa_supplicant.conf
root@beethoven:~# touch /mnt/ssh
root@beethoven:~# umount /mnt

> Also, your Pi didn't pick up date & time from the network. You might want to take a look at that, too.

I noticed that.  Later tracked it down to /data/etc/date.conf being empty.  I "fixed" it by copying /etc/date.conf which makes me now suspect something went wrong during the initial boot, when I'm guessing it sets up some configuration files under /data?.  This is what "fixed" ntp:  
cp -p /etc/date.conf /data/etc/date.conf

> maybe you don't have a proper gateway set?

DHCP worked O.K.  It connected to my wireless network on boot, I was able to ssh in, and could see the Internet:  

[root@meye-4ef5f9c2 ~]# ping google.com
PING google.com (216.58.213.14): 56 data bytes
64 bytes from 216.58.213.14: seq=0 ttl=116 time=13.222 ms
64 bytes from 216.58.213.14: seq=1 ttl=116 time=17.371 ms
^C

Which makes it very odd that it was apparently rebooting because it thought the network wasn't up, when in fact it was online.  

The second copy worked flawlessly.  Again sorry I didn't try making a new copy before taking your time.

closed 2022-01-30 by jjfelten

starbasessd, 2022-01-30:
No problem. One of the hazards of messing with read-only /boot and / partitions and file system overlays are you aren't sure which files are where.  anything in / is read only, including /etc. There are several overlay file systems which allow editing / changes. /data/etc is an example (as are /var/log, /var/lib & /usr)
If you do a df -h it shows the overlays and partition layout. 

Glad to see that you were able to get a second copy to work correctly. Enjoy.