motioneye-project/motioneyeos #2878: Raspi Zero2 W with Full HD

orcadiver, 2022-02-02:
Hi all,
my two Raspi Zero2 W are working fine so far.
Only the two highest resolutions (from 1600x1200) cause the Raspi to freeze (it requires a reboot), the highest (1920x1080) even irreversibly and requires the import of a new image - at any framerate, but at higher framerates it happens faster.
Can this be due to the performance of the Raspi or is it just the configuration.

starbasessd, 2022-02-02:
I vote configuration. With the Read-Only file system, it'd be interesting to see if / what gets corrupted. There are known issues with motion & higher resolutions and frame rates, OMX, and so forth. It's possible you are getting into 'race' conditions (where events start piling up, overlapping, unable to complete the previous trigger, etc...) I would switch to a non-OMX file format (highest GPU usage), lower resolution, lower frame rate, Movie Passthrough for 'recording files' (but you lose text overlays and such)...

starbasessd, 2022-02-02:
Is it possible that the /data partition gets full?
SSH into the Pi and run 
df -h
and report
Is it possible its overheating or has a bad (low voltage) power supply?
SSH into the Pi and run 
vcgencmd get_trhottled
and report

orcadiver, 2022-02-02:
a) it occurs (also) without motion detection
b) the 128GB SDCard is virtually empty

starbasessd, 2022-02-02:
Is it possible its overheating or has a bad (low voltage) power supply?
SSH into the Pi and run
vcgencmd get_throttled
and report

starbasessd, 2022-02-02:
Yeah, typo... Corrected...
vcgencmd get_throttled

orcadiver, 2022-02-02:
[root@motionEye-Cam1 ~]# vcgencmd get_throttled
throttled=0x0
[root@motionEye-Cam1 ~]#

starbasessd, 2022-02-02:
Try when it's frozen...

orcadiver, 2022-02-02:
just now the Raspi held for several seconds with 20 frames 1920x1080 (result of "vcgencmd get_throttled" continuously "throttled=0x0", then I switched to 25 frames and it froze (via SSH no reaction anymore)

starbasessd, 2022-02-02:
Interesting. Thoughts, @jawsper ?

orcadiver, 2022-02-02:
now it has been running for over a minute even at 1920x1080 and 30 frames, and gives the following message:
[root@motionEye-Cam1 ~]# vcgencmd get_throttled
throttled=0x20000
[root@motionEye-Cam1 ~]#

orcadiver, 2022-02-02:
...means undervoltage? ...with an original power supply with 5.1V and 2.5A?

starbasessd, 2022-02-02:
Yes, it can. What camera? USB or CSI? 
USB draw can be too much. CSI Camera can be defective. Power Supply can be defective. 
Try different Power Supply. If using an extender cable for power, take it out of the equation.

orcadiver, 2022-02-02:
CSI Raspi v2 (8M)

starbasessd, 2022-02-02:
https://rpi4cluster.com/scripts/undervoltage/ is a good script.
I have one I use:
nano /data/oops.py
- - - - -
#!/usr/bin/env python2

import subprocess

GET_THROTTLED_CMD = 'vcgencmd get_throttled'
MESSAGES = {
    0: 'Under-voltage!',
    1: 'ARM frequency capped!',
    2: 'Currently throttled!',
    3: 'Soft temperature limit active',
    16: 'Under-voltage has occurred since last reboot.',
    17: 'Throttling has occurred since last reboot.',
    18: 'ARM frequency capped has occurred since last reboot.',
    19: 'Soft temperature limit has occurred'
}

print("Checking for throttling issues since last reboot...")

throttled_output = subprocess.check_output(GET_THROTTLED_CMD, shell=True)
throttled_binary = bin(int(throttled_output.split('=')[1], 0))

warnings = 0
for position, message in MESSAGES.iteritems():
    # Check for the binary digits to be "on" for each warning message
    if len(throttled_binary) > position and throttled_binary[0 - position - 1] == '1':
        print(message)
        warnings += 1

if warnings == 0:
    print("Looking good!")
else:
    print("Houston, we may have a problem!")
- - - - -
Watches for both temp and volt, and when it caps.

orcadiver, 2022-02-02:
I have two identical combinations (raspi/camera/power supply) and with both the same effect.

starbasessd, 2022-02-02:
I don't know, but the CPU and its protections are having issues. 
Did you 'overclock' at all?

orcadiver, 2022-02-02:
no, everything works with the default parameters (except for the configuration by MotionOS)

starbasessd, 2022-02-02:
I don't know, but the CPU and its protections are having issues.

orcadiver, 2022-02-02:
with streaming qualiti 100% (uprated from default) now nothing works anymore

starbasessd, 2022-02-02:
It's not heat throttling. You can also check the messages log to see where the temps are...

orcadiver, 2022-02-02:
The backside of the board has about 50°C at the position of the CPU.

starbasessd, 2022-02-02:
vcgencmd measure_temp
or check the messaged log (Settings, Expert Settings)

orcadiver, 2022-02-02:
already borderline - or?

jawsper, 2022-02-02:
That is quite hot, mine is at 61.2 degrees with no active cooling or heatsink. I did leave the top off of the case because it feels like it runs a bit hotter than the Pi Zero (1) W.

starbasessd, 2022-02-02:
Beyond 'borderline'.

orcadiver, 2022-02-02:
Feb  2 16:27:17 motionEye-Cam1 user.notice throttlewatch: currently: arm-frequency-capped , temperature: 80 C
Feb  2 16:28:17 motionEye-Cam1 user.notice throttlewatch: currently: arm-frequency-capped , temperature: 80 C

starbasessd, 2022-02-02:
Most Pis start throttling at 60-65. Most will shut down at 85.

starbasessd, 2022-02-02:
Interesting that vcgencmd get_throttled doesn't indicate the heat issue. Maybe a change they made...

orcadiver, 2022-02-02:
then the good one without cooling is actually not suitable for this scenario - I wonder why raspi then markets such things

starbasessd, 2022-02-02:
They are marketed as educational tools. People take it and use it for other things.

orcadiver, 2022-02-02:
then the Raspi Zero 2W needs at least a heat sink!

starbasessd, 2022-02-02:
As my favorite YTer says: WATER COOLED!

orcadiver, 2022-02-02:
I can't quite agree with that - my config is more or less meant like this: Raspi Zero 2W with camera v2.1 and Raspi combi case

orcadiver, 2022-02-02:
[image]

orcadiver, 2022-02-02:
that does not serve its purpose then

starbasessd, 2022-02-02:
Go to RaspberryPi.org

starbasessd, 2022-02-02:
There were MAJOR issues with the Pi4Bs at the beginning with overheat, especially. They re-wrote a lot of firmware to lower the temps, and everyone came out with cooling solutions...

starbasessd, 2022-02-02:
Closing issue. Root Cause is heat.

closed 2022-02-02 by starbasessd