motioneye-project/motioneyeos #2879: Hi newbie here.

starbasessd, 2022-02-02:
Hi newbie here.

Currently setting up a Pi Zero 2 W with the current build provided by @jawsper (thanks for that!).

I am trying to deactivate the act_led but the howto from the wiki ain't working - the led stays active after reboot. There seems to be something inversed compared to the Pi Zero 1 W as the following switches the act_led off (the second echo should be 0 for the Pi Zero): echo none | tee /sys/class/leds/led0/trigger && echo 1 | tee /sys/class/leds/led0/brightness

dtparam=act_led_activelow=off instead of on did not work.

I already have another Pi Zero 1 W up and running were the howto from the wiki worked as described, motionEye 0.42.1 // Motion Version 4.2.2+gitUNKNOWN / OS Version motionEyeOS 20200606.

Tried to create a workaround but I am not familiar with linux and was not able to start crond on boot nor create a startup.sh under /etc/init.d/.

What am I missing to turn the act_led off via /boot/config.txt?

_Originally posted by @allpaulval in https://github.com/ccrisan/motioneyeos/issues/2829#issuecomment-1028417935_

starbasessd, 2022-02-02:
Did you try:
dtparam=act_led_trigger=none

starbasessd, 2022-02-02:
or

To make these settings permanent, add the following lines to your Pi's /boot/config.txt file and reboot:

# Disable the ACT LED on the Pi Zero.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on

allpaulval, 2022-02-02:
Thanks for the quick reaction but please do not stress, I am based in Germany and it's rather late.
Straight from my /boot/config.txt (like advised in the [HowTo ](https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs))
      # Disable the ACT LED.
      dtparam=act_led_trigger=none
      dtparam=act_led_activelow=on
ACT_LED is still going strong after the reboot ...

starbasessd, 2022-02-02:
I haven't been able to google anything specific for the PiZero2W. Only the other Pis. Maybe another user has ideas. Have you checked on the RPiFoundation User Forums?

allpaulval, 2022-02-02:
I searched too and found only someone with the same problem (who just used a workaround): [https://www.cnx-software.com/2021/12/09/raspberry-pi-zero-2-w-power-consumption/](https://www.cnx-software.com/2021/12/09/raspberry-pi-zero-2-w-power-consumption/) (search for "We can disable LED in theory by adding the following lines in config.txt")

For the suggested forum:
[Same Story - Workaround
](https://forums.raspberrypi.com/viewtopic.php?p=1957282&hilit=pi+zero+2+act_led#p1957282)

Gotta look more tomorrow, it's getting too late.
Have a great day!

allpaulval, 2022-02-04:
Okay, got some help and we created a workaround: "[comment] commands"

[remount for write permissions] mount -o remount,rw /
[create init.d-script] nano /etc/init.d/S100ACT_LED_OFF
[Line 1 script] #!/bin/bash
[Line 2 script] echo none > /sys/class/leds/led0/trigger
[Line 3 script] echo 0 > /sys/class/leds/led0/brightness
[set correct permissions] chmod 755 /etc/init.d/S100ACT_LED_OFF
[reboot] reboot

This does the trick and the led switches off after boot.

FYI:
1. I did remove the dtparam=act_led* in /boot/config.txt
2. I installed rasbian light and had the same problems. Couldn't find anyone solving this problem in their forum - just workarounds.