motioneye-project/motioneyeos #2880: Arducam 16MP with Raspberry Zero 2W

easyyu, 2022-02-06:
I am the owner of Raspberry 2W and ArduCam 16MP with sensor: Sony IMX519.
[https://www.arducam.com/16mp-autofocus-camera-for-raspberry-pi/](URL)
I installed MotionEyeOS from Jasper's latest version 20220119. But I was faced with the problem that MotionEyeOS can not find an MMAL camera :
[image]
[image]

Then I decided to download the latest image with Raspberry PI Imager and install manually the drivers from their web page:
[https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide/](url)

After driver installation I was able to make a still picture with the command:
`libcamera-still -t 5000 -o test.jpg --autofocus`

I installed MotionEye with help of a tutorial:
[https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](url)
with some additional help on how to install python2 on Bullseye.

But I run in the same problem even with installed drivers for ArduCam 16MP. I still get a message "No camera" under MMAL camera type.
Under /boot/config.txt I have a line:
`[all]
dtoverlay=imx519`

Commands of interest:
`root@raspberrypi:~# lsb_release -a                
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye`

`root@raspberrypi:~# v4l2-ctl --list-devices
unicam (platform:3f801000.csi):
        /dev/video0
        /dev/video1
        /dev/media3

bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/media2

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media1
`

`root@raspberrypi:~# v4l2-ctl -d /dev/video0 --all
Driver Info:
        Driver name      : unicam
        Card type        : unicam
        Bus info         : platform:3f801000.csi
        Driver version   : 5.10.92
        Capabilities     : 0x85a00001
                Video Capture
                Metadata Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : unicam
        Model            : unicam
        Serial           : 
        Bus info         : platform:3f801000.csi
        Media version    : 5.10.92
        Hardware revision: 0x00000000 (0)
        Driver version   : 5.10.92
Interface Info:
        ID               : 0x03000007
        Type             : V4L Video
Entity Info:
        ID               : 0x00000005 (5)
        Name             : unicam-image
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000006   : 0: Sink
          Link 0x02000009: from remote pad 0x1000002 of entity 'imx519 10-001a': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 0: ok)
Format Video Capture:
        Width/Height      : 4656/3496
        Pixel Format      : 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
        Field             : None
        Bytes per Line    : 5824
        Size Image        : 20360704
        Colorspace        : sRGB
        Transfer Function : sRGB
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Full Range
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 8, Top 48, Width 4656, Height 3496
        Default     : Left 8, Top 48, Width 4656, Height 3496
        Pixel Aspect: 1/1
Selection Video Capture: crop, Left 8, Top 48, Width 4656, Height 3496, Flags: 
Selection Video Capture: crop_default, Left 8, Top 48, Width 4656, Height 3496, Flags: 
Selection Video Capture: crop_bounds, Left 8, Top 48, Width 4656, Height 3496, Flags: 
Selection Video Capture: native_size, Left 0, Top 0, Width 4672, Height 3648, Flags: 

User Controls

                       exposure 0x00980911 (int)    : min=20 max=4028 step=1 default=1000 value=1000
                horizontal_flip 0x00980914 (bool)   : default=0 value=0 flags=modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=0 flags=modify-layout

Camera Controls

             camera_orientation 0x009a0922 (menu)   : min=0 max=2 default=2 value=2 flags=read-only
                                0: Front
                                1: Back
                                2: External
         camera_sensor_rotation 0x009a0923 (int)    : min=0 max=0 step=1 default=0 value=0 flags=read-only

Image Source Controls

              vertical_blanking 0x009e0901 (int)    : min=564 max=8380504 step=1 default=564 value=564
            horizontal_blanking 0x009e0902 (int)    : min=12240 max=12240 step=1 default=12240 value=12240 flags=read-only
                  analogue_gain 0x009e0903 (int)    : min=0 max=960 step=1 default=0 value=0
                red_pixel_value 0x009e0904 (int)    : min=0 max=4095 step=1 default=4095 value=4095
          green_red_pixel_value 0x009e0905 (int)    : min=0 max=4095 step=1 default=4095 value=4095
               blue_pixel_value 0x009e0906 (int)    : min=0 max=4095 step=1 default=4095 value=4095
         green_blue_pixel_value 0x009e0907 (int)    : min=0 max=4095 step=1 default=4095 value=4095

Image Processing Controls

                     pixel_rate 0x009f0902 (int64)  : min=686000000 max=686000000 step=1 default=686000000 value=686000000 flags=read-only
                   test_pattern 0x009f0903 (menu)   : min=0 max=4 default=0 value=0
                                0: Disabled
                                1: Color Bars
                                2: Solid Color
                                3: Grey Color Bars
                                4: PN9
                   digital_gain 0x009f0905 (int)    : min=256 max=65535 step=1 default=256 value=256`

Is it possible to run the MotionEye project with this camera?
Any help will be appreciated.

starbasessd, 2022-02-06:
In 
sudo raspi-config
do you have 
Interfaces, Camera Interface, Enabled=Yes?

starbasessd, 2022-02-06:
You also need to use the correct instructions for [Bullseye](https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye) if you are using libcamera....

easyyu, 2022-02-07:
Hi,
First of all, thank you for your help.
I reinstalled the Raspberry OS and follow the instruction for Bullseye, but it is the same. It points to the same deb.
Yes, I tried to enable over raspi-config the camera, but after reboot, I still have not camera in MotionEye MMAL menu.
With the enabled camera I can not use anymore the command:
`libcamera-still -t 5000 -o test.jpg --autofocus`
because I got the error:
```
`root@raspberrypi:~# libcamera-still -t 5000 -o test.jpg --autofocus
Preview window unavailable
[0:00:57.665455602] [1153]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:00:57.713280393] [1155]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx519'
[0:00:57.713673987] [1155]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:00:58.263316747] [1155] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/video0
[0:00:58.271536226] [1153]  INFO Camera camera.cpp:937 configuring streams: (0) 2328x1748-YUV420
[0:00:58.273053883] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt RG10 Score: 2519 (best 2519)
[0:00:58.273304299] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt RG10 Score: 2148.24 (best 2148.24)
[0:00:58.273452633] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG10 Score: 1500 (best 1500)
[0:00:58.273622685] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 3819.24 (best 1500)
[0:00:58.274004872] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG10 Score: 5819.24 (best 1500)
[0:00:58.274208310] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt pRAA Score: 2019 (best 1500)
[0:00:58.274388831] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt pRAA Score: 1648.24 (best 1500)
[0:00:58.274549976] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pRAA Score: 1000 (best 1000)
[0:00:58.274865654] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 3319.24 (best 1000)
[0:00:58.275014976] [1155]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRAA Score: 5319.24 (best 1000)
[0:00:58.275646747] [1155]  INFO RPI raspberrypi.cpp:624 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected mode: 2328x1748-pRAA
[0:00:58.428923518] [1155] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video14[15:cap]: Not enough buffers provided by V4L2VideoDevice
[0:00:58.431818466] [1155] ERROR RPI raspberrypi.cpp:821 Failed to allocate buffers`
```

Of course, I changed the: **gpu_mem** from **64** to **128** and later even for higher value but without success.

Do you have any suggestion?

starbasessd, 2022-02-07:
OK, let's start over.
Install RaspiOS [Buster](https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2021-12-02/2021-12-02-raspios-buster-armhf-lite.zip)
>sudo apt update && sudo apt upgrade -y

Set up as needed in raspi-config (including enable CSI Camera in interfaces):
**DO NOT install the libcamera packages. They disable and break the interfaces motionEye uses.**
>sudo raspi-config
>reboot

Set up motionEye using instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian):
>sudo su
>apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y
>wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb
>dpkg -i pi_buster_motion_4.3.2-1_armhf.deb
>apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev -y
>pip install motioneye
> mkdir -p /etc/motioneye
>cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
>mkdir -p /var/lib/motioneye
>cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
>systemctl daemon-reload
>systemctl enable motioneye
>systemctl start motioneye
>reboot

Connect to the http://ip_adddress_of_Pi:8765 and see if your camera shows (It should be already discovered and set up, but may need to be manually set up)

If it still doesn't show, I don't know what to tell you.
Check for camera discovery with command:
>vcgencmd get_camera

Should show:
supported=1 detected=1 if the camera is correctly connected.

easyyu, 2022-02-08:
OK, my fault, I didn't pay attention to the "pillow" package and warning for 99% in the Orig tutorial. I solved with:
`apt-get install python-pil -y `
and re-run:
`pip install motioneye`

In the end I got MotionEye running by service, and after the restart, I still have no MMAL camera :(

Command **vcgencmd get_camera** giving me:
```
root@raspberrypi:~# vcgencmd get_camera
supported=1 detected=0
```

If I install the original drivers for the camera I can use it and I can get the pictures, which proves that camera is usable.

starbasessd, 2022-02-08:
After googling the 'vcgencmd get_camera supported=1 detected=0' error and your camera specifically, it appears that it isn't supported in legacy Buster, only the 'new' Buster with the updated 'picamera' libraries and missing python support (the 'drivers' you are referring to). This will prevent it from working with motionEye or motionEyeOS. Previous versions of Arducam (12MP and earlier) are/were supported in Legacy Buster. MMAL cameras in Bullseye and motionEye are also not supported due to this same issue. You can use USB and Network cameras in both Buster and Bullseye motionEye.

starbasessd, 2022-02-08:
Some references:
https://github.com/raspberrypi/linux/issues/4697
https://www.arducam.com/docs/cameras-for-raspberry-pi/native-raspberry-pi-cameras/faq/
https://duckduckgo.com/?q=vcgencmd+get_camera+supported%3D1+detected%3D0+%22Arducam+16MP%22&t=ffab&ia=web

closed 2022-02-09 by starbasessd

wickerman123, 2022-06-01:
Has there been any updates in this area? I just got an Arducam 16mp NOIR and I'm having the same issues - with MotionEyeOS on an Pi 4 and Pi Zero 2W.

starbasessd, 2022-06-01:
Pi4 kernel is still pre-16MP (dev20201026) I don't know if @jawsper updated the kernel when he created his PiZero2W image or not.
You can install the latest Legacy Buster and it _should_ work, then install motionEye from the [motionEye](https://github.com/motioneye-project/motioneye) github. That has the advantage of being all Python3 code.